OctoPrint/OctoPrint #3490: get_template_configs can't replace gcodeviewer

jneilliii, 2020-03-21:
### What were you doing?

Trying to modify the PrettyGCode plugin to be able to replace the GCode Viewer tab using the get_template_configs callback. You can find the specific item related right [here](https://github.com/jneilliii/OctoPrint-PrettyGCode/blob/347cb6b51369ac66ec2b66dca84eb6a30cc2cd0e/octoprint_prettygcode/__init__.py#L22).

#### What did you expect to happen?

For the PrettyGCode tab to take the place of the default GCode Viewer tab as described [here](https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.TemplatePlugin.get_template_configs) in the docs using `replaces`

#### What happened instead?

Both the default tab and the PrettyGCode tab still load together, the replacement doesn't occur.

#### Did the same happen when running OctoPrint in safe mode?

N/A can't reproduce without plugin loading.

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

Octopi 0.17 and Windows 10 Development environment.

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 80.0.3987.149 (Official Build) (64-bit)

#### Link to octoprint.log

N/A, no relavant errors in log.

#### Link to contents of terminal tab or serial.log

N/A 

#### Link to contents of Javascript console in the browser

N/A No relevant errors in developer console.

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2020-04-03:
That's a bug in the plugin, `get_template_config` instead of `get_template_configs`. Should be this:

``` python
    def get_template_configs(self):
        return [
            dict(type="tab", template="prettygcode_tab.jinja2", replaces="gcodeviewer", name="GCode Viewer"),
            dict(type="generic", template="prettygcode.jinja2")
        ]
```

closed 2020-04-03 by foosel

loelkes, 2020-08-11:
In 1.4.2 I can replace every tab (tested with _temperature_ and _control_) but not the _gcodeviewer_. Did something change here, like the name of the tab?

foosel, 2020-08-11:
Yep, the gcode viewer was moved into a bundled plugin and hence it should now be `plugin_gcodeviewer`. I just noticed I also forgot to update the [(appearance config) docs](https://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance) with that where that stuff can be found, and just fixed this. Docs build is running at the moment, should be up shortly.