OctoPrint/OctoPrint #3491: [Request] FileMoved Event

jneilliii, 2020-03-22:
**Is your feature request related to a problem? Please describe.**
With my UltimakerFormatPackage plugin images are extracted to the plugin's data folder and these images get removed when the FileRemoved event is triggered. This event  is triggered on File move as well based on the current implementation and the details in the original request [here](https://github.com/foosel/OctoPrint/issues/1092). This in turn removes the image from the plugin's data folder, but there is no way to identify if the FileRemoved event is because of a move or a delete operation. So in the situation where the file is moved by either the [FileManager](https://plugins.octoprint.org/plugins/filemanager/) or the [DeleteMoveAfterPrint](https://plugins.octoprint.org/plugins/DeleteAfterPrint/) plugins the link to the image gets broken because the image file gets removed.

**Describe the solution you'd like**
Either an additional FileMoved event, or properties added to the existing FileAdded/FileRemoved events to indicate a file move versus a simple delete.

**Describe alternatives you've considered**
The only workaround I can think of on my side is to create a delayed timer on FileRemoved event to delay the image removal after the FileRemoved event is triggered that gets cancelled on the FileAdded event if the flagged name is the same and then moves the image instead of deleting it. Timing on that would need to be verified though, is there ever an overlap of the two existing event triggers? Only other issue I can think of related to that delayed timer is if a file gets overwrote because of an upload with the same name. Does the FileRemoved event get triggered in those situations as well or just the FIleAdded event?

**Additional context**
Issues posted [here](https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/issues/10) and [here](https://github.com/OllisGit/OctoPrint-DeleteAfterPrint/issues/11).

kantlivelong, 2020-04-13:
I agree, there should be a `FileMoved` and `FolderMoved` event. It is easy enough to implement but the concern is backwards compatibility. If the `FileAdded`, `FileRemoved`, `FolderAdded`, `FolderRemoved` events are replaced with `FileMoved` and `FolderMoved` then some plugins won't get the update at all. If they are added in addition to then that could cause an issue with an action occurring twice (maybe?). 

I've pushed [https://github.com/kantlivelong/OctoPrint/commit/72ad970cd6b2534347465188fa88a9b1197b0498](https://github.com/kantlivelong/OctoPrint/commit/72ad970cd6b2534347465188fa88a9b1197b0498) to my fork which could be introduced as a PR if @foosel approves. I haven't yet tested it though.

jneilliii, 2020-04-13:
For my specific use case, as long as the move event happened prior to the add/remove events then I could more easily program around that rather than keeping an array of the files removed/added and using a threaded timer to do the work later and it wouldn't effect other plugins. Maybe add a deprecation warning for the add/remove events on move and take those out at a later release?

kantlivelong, 2020-04-13:
That's what I figured. 

Move -> Remove -> Add

I can't think of any case where a file being removed twice would cause a plugin problems unless they are written poorly. Though the double action is still kinda _meh_. 

Unfortunately, I don't believe there is a way to provide a deprecation message for events. as they are all called with same fire_event() method and plugins listening for events just check for a value.

jneilliii, 2020-04-13:
> Unfortunately, I don't believe there is a way to provide a deprecation message for events. as they are all called with same fire_event() method and plugins listening for events just check for a value.

You could log it though from within the move function and then add a note to the documentation [here](https://docs.octoprint.org/en/master/events/index.html#file-handling), assuming @foosel agrees. The only plugin authors I know of that could potentially be using these or benefit from are @OllisGit and @Salandora.

kantlivelong, 2020-04-13:
That would be called for everything though not just specific plugins using deprecated functionality.

foosel, 2020-04-14:
I would merge a PR that adds `*Moved` events before the existing ones and updates the documentation to mark the old ones as deprecated in case of move (they are still relevant for regular add/remove operations outside of moves). What about `*Copy`? That's currently not distinguishable from file/folder addition either. If we touch this stuff for move I think we should also touch it for copy.

About deprecation warnings, I guess it's time to add a new section to release notes for stuff like that, to highlight it. And maybe section in the docs or on the forum with current deprecations as well? Trying to make this stuff easier to discover but I'm a bit worried about keeping it synced.

kantlivelong, 2020-04-14:
Currently `copy_folder` simply fires a FolderAdded event but does not fire events for sub-folders or files. Should this be made recursive to fire events per sub folder/file?

jneilliii, 2020-06-13:
@kantlivelong I think the plugins could handle the recursion themselves if they knew of a `FolderCopied` event. For the purposes of the plugins that initiated this request it would just be a matter of copying the associated files in the plugin's data folder and then recursively update the underlying file's metadata, so I think consistency between the `FolderMoved` and `FolderCopied` events would be the most important thing.

@foosel It would be nice to have a separate "plugin developers" announcement channel to be able to give developers a heads up way in advance of release (ie on push/commit of a deprecation flag of some sort) to give them time to prepare. I'm sure you could figure out some way to integrate that into the current github actions somehow and then on release maybe merge the announcement channel into the release notes for you in order for those to stay in sync?