OctoPrint/OctoPrint #3492: version api returns 200 OK (+ data) without any x-api-key

UnchartedBull, 2020-03-22:
#### What were you doing?

1. Send an API request to the version API without any authentication (no x-api-key)
2. Get the version information

#### What did you expect to happen?
In version 1.3.x the version api did return a 403 unauthorised response, with 1.4.0 I'm getting 200 ok. I didn't find anything, that mentioned changing this and also the docs are still listing the api-key in the example request.

I don't know if this is wanted behaviour or not. I did change nothing in the ACL settings, so the default Guest user isn't allowed anything. Please let me know if this is desired the behaviour for OctoPrint going onwards from 1.4.0. The version API always was a great way to test if a token is valid for my app. I did test other api points (as seen in the logs), which return the correct 403 status. 

#### What happened instead?
Got a successful response instead of an error one

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPrint
1.4.0

#### Operating System running OctoPrint
Debian 10

#### Printer model & used firmware incl. version
Ender-3 Pro with Marlin 2.0

#### Browser and version of browser, operating system running browser
Chrome, macOS & Windows

#### Link to octoprint.log
https://pastebin.com/vjryUC91

#### Link to contents of terminal tab or serial.log
n.a.

#### Link to contents of Javascript console in the browser
n.a.

#### Screenshot(s)/video(s) showing the problem:
[image]

headers are only the standard postman headers, no x-api-key

I have read the FAQ.

foosel, 2020-04-24:
That was an oversight and should be solved by the above commit. Fix ready for 1.4.1.

UnchartedBull, 2020-04-24:
Ok, thanks. Closing this, as it should be fixed.

commit 0a889dc7 (2020-04-24): 🐛 Fix permissions on /api/status

Now requires the STATUS permission.

Fixes #3492
closed 2020-04-24 by UnchartedBull