OctoPrint/OctoPrint #3496: [Request]: Add Restore Option on Initial Install Wizard

jneilliii, 2020-03-28:
**Is your feature request related to a problem? Please describe.**
Not specifically a problem, but an inconvenience. When initially flashing an SD card with OctoPi you have to go complete the initial setup wizard before you can get to the point where you can restore. This is counter-intuitive when the end goal is to restore from a backup anyway.

**Describe the solution you'd like**
Inject a step in the initial configuration wizard that precedes all others giving the option to restore from backup, therefore voiding the remaining steps.

**Describe alternatives you've considered**
Possibly an initial prompt that asks do you want to restore from backup and configure from scratch.

**Additional context**
As discussed on the last OctoPrint On-Air #30, creating official feature request to remind you @foosel.

kantlivelong, 2020-03-31:
So I took a look at this and whipped up a little test w/ the CoreWizard plugin. I think these would be considered blockers.

1. I don't see any way to set the order of SubWizards. Fairly important as this would need to be the first step.
2. The restore step in the wizard would need to either disable ACLs or use a random value prior to restoring otherwise 403 gets returned on upload.

I've got a working proof of concept @ https://github.com/kantlivelong/OctoPrint/commit/b63eec3e919edb618f06ae192b2a80e686d94d1b

@foosel thoughts?

foosel, 2020-04-01:
> * I don't see any way to set the order of SubWizards. Fairly important as this would need to be the first step.

That can actually be done by adjusting the (default) values for `appearance.components.wizard` in `settings.py` to `["plugin_corewizard_backuprestore", "plugin_corewizard_acl"]` (side note: currently it says `access` and that's a left-over from before stuff was extracted into the corewizard plugin, I'm surprised this hasn't caused issues).

> * The restore step in the wizard would need to either disable ACLs or use a random value prior to restoring otherwise 403 gets returned on upload.

I'd say we make the wizard not a part of the corewizard plugin but of the backup plugin itself and adjust the backup plugin's upload endpoint to take ACLs into account. If you don't mind I'm going to experiment a bit with that since I kinda need something possibly interesting to look at for the livestream in half an hour and I'm kinda dug into this code now anyhow.

kantlivelong, 2020-04-01:
> That can actually be done by adjusting the (default) values for appearance.components.wizard in settings.py to ["plugin_corewizard_backuprestore", "plugin_corewizard_acl"] (side note: currently it [...]

Ah! 

> I'd say we make the wizard not a part of the corewizard plugin but of the backup plugin itself and adjust the backup plugin's upload endpoint to take ACLs into account. If you don't mind I'm going [...]

Awesome!

foosel, 2020-04-01:
Implemented by the above commit and some more, ready for 1.4.1 (and the first feature to be finalized in a live stream 😅)

kantlivelong, 2020-04-01:
:tada: :tada: :tada: 

[Link to stream](https://www.youtube.com/watch?v=Dm1xpUMkrRY) for anyone interested.

jneilliii, 2020-04-01:
Great live feed Gina.

commit 7e4c29be (2020-04-01): ✨ backup: Add first run wizard

Bypasses permissions in first run mode, allows
to restore existing backup.

Closes #3496
closed 2020-08-04 by foosel