OctoPrint/OctoPrint #3498: [Request] Show popup message to support makervsvirus.org

od-git, 2020-03-29:
If OctoPrint could show a news popup in its user interface to point people to https://www.makervsvirus.org/, this could very specifically communicate the idea to many people who could use their 3D printers to help the maker movement fight the corona pandemic.

od-git, 2020-03-29:
Please note that this initiative is currently only active in German speaking countries, although there is an English version of the website.

overflo23, 2020-03-29:
I really like this idea, there are >500 people fro germany alone in the #slack channel.
Lets help out.

od-git, 2020-03-29:
> I really like this idea, there are >500 people fro germany alone in the #slack channel.
> Lets help out.

Its actually already >1800

foosel, 2020-03-30:
I've been giving this a lot of thought over the past 14h and I have to admit that I feel uncomfortable using the announcements or rather the blog (because that's what powers the announcements really) as a means to push people *globally* towards one currently very *regional* coordination platform.

But I can also see how this could indeed be helpful. I think the best way forward might be with a guest blog post. If someone on here would commit to writing a post that highlights *a number* of ways people can currently coordinate to help with their 3d printers in the fight against corona, I'll happily publish that (and thus push it to all the instances out there as an announcement as well). 

I'm thinking something like makervsvirus.org, masksfordocs.com, even more regional coordination efforts I might not have gotten wind yet. So something that really gives almost everyone worldwide who'd read that an option to help, not go "oh, just Germany, too bad :("

Anyone up for that?

od-git, 2020-03-30:
Thank you for your feedback. I totally agree. 

I've forwared this to the PR orga. See https://www.makervsvirus.org/press/

od-git, 2020-03-31:
Now below is my (rough) proposal. There are people on makervsvirus and masksfordocs who eventually will further complement / correct this text.

# Fight corona crisis using your 3D printer

During the last days lots of projects appeared worldwide, all aiming to coordinate makers with 3D printers (and laser cutters etc.) with the drastically rising demand for medical personal protection equipment (PPE).

Now people all over the world are designing and printing:
 * Protective Faceshields (https://www.prusaprinters.org/prints/25857-protective-face-shield-rc2), 
 * Open Source Mask (https://www.opensourcemask.com/en/), 
 * Hands-Free Door Openers (https://www.materialise.com/en/hands-free-door-opener)
 * etc.

If you also like to contribute, you should connect to a local coordination hub through one of the below listed websites. There you can match your own production capability with the current local demand. This is important, because most medicals are asking specifically for something like 70 faceshields of a certain kind but not for a surprise kit. Most projects communicate using Slack or Telegram.

Example: Many hackerspaces did become local coordination hubs and are now aiming to produce a certain amount of Prusa Face Shield RC3 which have been requested by nearby surgeries and hospitals. These hackerspaces run 3D printers and lasercutters by their own but as well collect home printed parts from many nearby makers and companies. 

Even if you probably can not produce all of the parts on your own (like lasercutting the foil) you can contribute substantially by delivering demanded parts to the coordinating hubs. They will complement remaining parts, store them and finally deliver them. 

If you join the movement, please remember to treat the parts as if you were already infected: Use protective gloves and masks when handling the parts and please follow the instructions of your coordinating hub.

Here is a list of some larger coordination projects by location:

**International**
 * masksfordocs.org
 * getusppe.org/makers/
 * www.helpfulengineering.org/volunteer/

**Germany**
 * makervsvirus.org
 * medprint.org
 * print2protect (https://info743967.typeform.com/to/o8l610)

**UK**
 * national3dprintingsociety.co.uk (some US as well)

**Spain**
 * https://www.coronavirusmakers.org/

_Please note: This list is not complete for sure. So please feel free to add more projects in the comments, if something is missing._

foosel, 2020-03-31:
That looks good! Would leave out mask designs though - the verdict on the safety and actual effectiveness of these things is still out and I'd prefer to stick to proven ways to help. 

Do we have an ETA on a final version - since you mentioned further comments/revisions were likely incoming? Or did I misunderstand that?

od-git, 2020-03-31:
> Would leave out mask designs though - the verdict on the safety and actual effectiveness of these things is still out and I'd prefer to stick to proven ways to help.

I agree

> Do we have an ETA on a final version - since you mentioned further comments/revisions were likely incoming? Or did I misunderstand that?

I think we don't have to wait ages for that. These people are offline currently or can do a review earliest in the afternoon (CET). I'll just start a last RFC in slack right now.

AlexMammut, 2020-03-31:
Fine with me (press/communication manager @ makervsvirus.org )

foosel, 2020-03-31:
I cleaned up the proposal a bit, set links as needed and also added a small "this is a guest post" tidbit on top. I hope it's ok for you @od-git that I'll set you as the author.

---

# Help to fight the corona crisis using your 3D printer

*A guest post by od-git on how you can help in the fight against the ongoing SARS-CoV-2 pandemic. If you have any printing capacities free, please consider joining the community effort! Thank you. ~ Gina*

During the last days a lot of projects appeared worldwide, all aiming to coordinate makers with 3D printers (and laser cutters etc.) with the drastically rising demand for medical personal protection equipment (PPE).

Now people all over the world are designing and printing:
 * [Protective Faceshields](https://www.prusaprinters.org/prints/25857-protective-face-shield-rc2), 
 * [Hands-Free Door Openers](https://www.materialise.com/en/hands-free-door-opener)
 * etc.

If you also would like to contribute, you should connect to a local coordination hub through one of the below listed websites. There you can match your own production capability with the current local demand. This is important, because most medicals are asking specifically for something like 70 faceshields of a certain kind but not for a surprise kit. Most projects communicate using [Slack](https://slack.com) or [Telegram](https://telegram.org/).

Example: Many hackerspaces did become local coordination hubs and are now aiming to produce a certain amount of Prusa Face Shield RC3 which have been requested by nearby surgeries and hospitals. These hackerspaces run 3D printers and lasercutters by their own but as well collect home printed parts from many nearby makers and companies. 

Even if you probably can not produce all of the parts on your own (like lasercutting the foil) you can contribute substantially by delivering demanded parts to the coordinating hubs. They will complement remaining parts, store them and finally deliver them. 

If you join the movement, please remember to treat the parts as if you were already infected: Use protective gloves and masks when handling the parts and please follow the instructions of your coordinating hub.

Here is a list of some larger coordination projects by location:

**International**

 * [#MaskForDocs](https://masksfordocs.com/)
 * [#GetUsPPE](https://getusppe.org/makers/)
 * [Helpful Engineering](https://www.helpfulengineering.org/volunteer/)

**Germany**

 * [#MakerVsVirus](https://www.makervsvirus.org/)
 * [MedPrint](http://medprint.org/)
 * [print2protect](https://info743967.typeform.com/to/o8l610)

**UK**

 * [National 3D Printing Society](https://national3dprintingsociety.co.uk/) (some US as well)

**Spain**

 * [Coronavirus Makers](https://www.coronavirusmakers.org/)

_Please note: This list is not complete for sure. So please feel free to add more projects in the comments, if something is missing._

od-git, 2020-03-31:
These hackerspaces run 3D printers and lasercutters **~~by~~ on** their own but as well collect home printed parts from many nearby makers and companies.

foosel, 2020-03-31:
Adjusted

foosel, 2020-03-31:
Blog post prepared, including a quickly fiddled together face shield for the octopus. Final go as reply to this please :)

[image]

od-git, 2020-03-31:
Great, I think we're done. I like the logo! Lets go. And thousand thanks!!

AlexMammut, 2020-03-31:
Go! I love the octopus with face shield!

foosel, 2020-03-31:
And we are live: https://octoprint.org/blog/2020/03/31/3d-printing-against-corona/

[image]

closed 2020-03-31 by foosel