OctoPrint/OctoPrint #3500: No recognition of tool after successful job completion

throwaway1349, 2020-03-31:
#### What were you doing?

1. Print arbitrary file with tool 1
2. Wait for it to finish - all is fine.
3. Start print job a second time.

#### What did you expect to happen?
successful print start and completion
#### What happened instead?
No recognition of used tool. OctoPrint displays a temperature (205°C) but refuses to print: "cold extrude prevented"
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
OctoPrint 1.3.12

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
Model: Multirap M420_2Move_12
Firmware 2.1.0

#### Browser and version of browser, operating system running browser
unsure

#### Link to octoprint.log

https://gist.github.com/throwaway1349/ccf35473c9ddfbd9ddd6fdd95b6ac743
#### Link to contents of terminal tab or serial.log
serial.log in same gist as above

Tool in question is heated
[image]

Tool in question supposedly doesn't exist
[image]

Tool in question shall not extrude
[image]

I have read the FAQ.

foosel, 2020-03-31:
> OctoPrint displays a temperature (205°C) but refuses to print: "cold extrude prevented"

That's not OctoPrint refusing to print, that's your firmware refusing to print.

> serial.log in same gist as above

The log is sadly incomplete, it only contains an endless stream of cold extrusion errors from the firmware, but I need to know what happened before, especially what happened on print end of the successful first print that led to the tool being reported as non existent.

Also, just in case, please make sure you actually have configured multiple tools in your printer profile, even if they all share a nozzle:

[image]

Also this is concerning and also makes me question whether you actually ran in safe mode because at least that restart into safe mode completely failed:

```
2020-03-30 15:06:43,558 - octoprint.server.api.system - WARNING - Command for core:restart_safe failed with return code 139:
! STDOUT: 
! STDERR: Segmentation fault
! Segmentation fault
```

As is this:

```
2020-03-30 15:46:24,115 - tornado.application - ERROR - Uncaught exception GET /downloads/logs/serial.log (::ffff:10.90.90.95)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1512, in _execute
    result = yield result
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
    value = future.result()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1069, in run
    yielded = self.gen.send(value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2422, in get
    yield self.flush()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 953, in flush
    return self.request.connection.write(chunk, callback=callback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 437, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 412, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
```

Ever pulled the power out of the Pi before shutting it down properly by any chance?

throwaway1349, 2020-04-03:
> The log is sadly incomplete, it only contains an endless stream of cold extrusion errors from the firmware, but I need to know what happened before, especially what happened on print end of the successful first print that led to the tool being reported as non existent.

throwaway1349, 2020-04-03:
Hey, thank you for your answer.

> The log is sadly incomplete, it only contains an endless stream of cold extrusion errors from the firmware, but I need to know what happened before, especially what happened on print end of the successful first print that led to the tool being reported as non existent.

Here is the code of the end of the first print. I detect that the error is already here at the end of the print. And not only for Extruder T1 also for Extruder T0.

[image]

> Also, just in case, please make sure you actually have configured multiple tools in your printer profile, even if they all share a nozzle:

The configuration of the printer profile are correct. 

> Also this is concerning and also makes me question whether you actually ran in safe mode because at least that restart into safe mode completely failed:

I restartet the Pi a second time correctly in safemode. But the problem is still there.

> Ever pulled the power out of the Pi before shutting it down properly by any chance?

Yes, it happened some times.

foosel, 2020-04-03:
I need a full serial.log. Not excerpts or screenshots, the full thing from initial connection to the printer until error and end.

foosel, 2020-04-03:
The message from your printer there that goes "M104 Invalid extruder \x03" looks corrupted btw. It should have a tool identifier there, not a binary value. Also your printer seems to think you don't have 4 tools, so make sure your slicer knows about that as well because that M104 T3 command there seems to be what triggers the incorrectly formatted error coming from the firmware.

throwaway1349, 2020-04-03:
> The message from your printer there that goes "M104 Invalid extruder \x03" looks corrupted btw. It should have a tool identifier there, not a binary value. Also your printer seems to think you don't [...]

The problem is solved!
In the printer profile, which the printer manufacturer provides for Simplify3D, the nozzles T2 and T3 are cut off at the end of printing. Although they are not there. After removing these lines in the script, the error has disappeared. The strange thing is that it worked a few weeks ago, even with these lines.

closed 2020-04-03 by throwaway1349