OctoPrint/OctoPrint #3501: Cannot instal plugin after fresh install

thomasblc, 2020-03-31:
Hi everyone!

I installed octoprint on my raspberry pi 3B+ and i am having issue to install any plugin from the plugin manager.
**What I want to do**: Install any 3rd party software

**What's wrong**: from the plugin manaer, I have a error message at the step: "building wheel for numpy (setup.py): started "

It looks like it may be an issue with pillow because the plugin manager has been stucked several time at the step "collecting pillow" and then it stopped doing anything and the server got disconnected.

I tried to connect via SSL and do the following :
_source oprint/bin/activate_
_pip install --upgrade pip_ --> worked until here
_pip install pillow --no-cache-dir_ --> didn't work and the server stopped responding.

I did'nt do anything with my raspberry and octoprint so I think my set-up is still clean and I don't know what causes this issue or how to resolve it.

I would be really glad if you could advise me on what to do next so I can finally uses octoprint to its full potential with the plugins :)

Thanks in advance !

thomasblc, 2020-03-31:
[plugin_pluginmanager_console (1).log](https://github.com/OctoPrint/OctoPrint/files/4410591/plugin_pluginmanager_console.1.log)
[image]
[image]

thomasblc, 2020-03-31:
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/4410595/octoprint.log)

foosel, 2020-03-31:
That looks like an issue with your environment and not OctoPrint, please open a thread in the community forum at https://community.octoprint.org. Thank you.

closed 2020-03-31 by foosel