OctoPrint/OctoPrint #3502: Loading Failed: Web interface doesn't load on 1.4.0

tanah, 2020-04-01:
#### What were you doing?

After updating Octoprint from 1.3.7 to 1.4.0 on my OctoPi 0.17.0 image the web interface doesn't load. A 'Loading Failed' message pops up and refers to the browser logs and octoprint.log file but I don't know how to identify the problem. By the looks of it it has something to do with socket connection.

I tried reflashing the SD card with a clean installation. I have no third party plugins installed, only klipper installed prior to reflashing the SD card. I also used safe mode just in case to no avail.

Here are the steps I followed to reproduce the error.

1. Burn SD card with OctoPi 0.17.0 using Etcher
2. Start the RPi, enter octopi.local and complete the installation wizard without changing the default configuration
3. Click on the update to 1.4.0 version of Octoprint
4. Click on Restart server after the update is finished
5. Enter octopi.local to find the Loading Failed screen (both in normal browser mode and private mode in case is about cache)
6. Enter the device through ssh and used sudo apt-get update and sudo apt-get upgrade just in case
7. Edited the config.yaml file to start in safe mode just in case
8. Restart octoprint with sudo service octoprint restart
9. Same result as step 5

The network tab in the browser console doesn't show stuck processes.

My current hardware consists of a Raspberry PI B+ v1.2 connected to the internet through ethernet cable with a double power supply (split USB cable) of 5V 3A nominal each.

#### What did you expect to happen?

See the usual web interface in octopi.local

#### What happened instead?

Loading Failed screen, no web interface could be loaded.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

No printer connected, so not applicable

#### Browser and version of browser, operating system running browser

Firefox 74.0
5.5.13-1-MANJARO

#### Link to octoprint.log

[octoprint.log](https://gist.github.com/tanah/8dfa9cf175cbb092be062da4209ad6da)

#### Link to contents of terminal tab or serial.log

[terminal tab](https://gist.github.com/tanah/51e40182da8587b6fdb930b2c169f6a7)

#### Link to contents of Javascript console in the browser

[javascript console](https://gist.github.com/tanah/1878509dedb1057e6adb4ec34b385ea2)

#### Screenshot(s)/video(s) showing the problem:

[Screenshot](https://imgur.com/dGs137q)

I have read the FAQ.

N3WWN, 2020-04-03:
I am experiencing the same problem. 

Doesn't matter what browser I use (Chrome, Safari, Firefox) or device (MacOS X, iPad, Linux, Android), 95% of the time, I get this error:

[image]

I've tracked it down to what appears to be something hanging in octoprint.server.util.sockjs.

octoprint.log shows a large delay (usually 10-30+ seconds) between the previous line logged and `octoprint.server.util.sockjs - INFO - New connection from client: xxx.xxx.xxx.xxx`.  

It is during the delay duration that the web browser times out and shows the error.    Then the log displays the sockjs line.  

Successful attempts have the sockjs line show up in the logs only a few seconds after the previous line.  I keep referring the 'the previous line' because the particular line changes.

All logging is set to debug level.

I thought it may be a DNS resolution problem, so made sure DNS was configured properly on my CentOS 7 octoprint host and that my client address resolves by adding an entry in /etc/hosts.

My setup is a fresh install of OctoPrint 1.4.0:

```
2020-04-01 14:52:40,604 - octoprint.environment - INFO - Detected environment is Python 2.7.17 under Linux (linux2). Details:
|  hardware:
|      cores: 16
|      freq: 0.0
|      ram: 33550057472
|  os:
|      id: linux
|      platform: linux2
|  python:
|      pip: 20.0.2
|      version: 2.7.17
|      virtualenv: /home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint
```

Restarting OctoPrint often increases the probability of being able to pull up the interface, but there's still a 30-60% chance of getting the 'Loading Failed' error.

Let me know if there's anything I can do to help troubleshoot or resolve this!

Thanks!

-Rich

foosel, 2020-04-03:
I have to admit, at the moment I'm at a complete loss, and I also haven't seen it happening myself. 

What I can do is increase the timeout on the `maintenance` branch and someone who can reproduce it can then check if that at least solves the symptom. I have no idea what could be taking so long in there, especially not under safe mode.

foosel, 2020-04-03:
One thing, can both of you please check the Network tab in the developer console of the browser and tell me what sockjs endpoints are being used? (See also https://webmasters.stackexchange.com/a/77337)

[image]

There should normally be an open connection to something that looks like this:

    /sockjs/283/vn2flmnd/websocket

It might also be a different transport than `websocket` however (e.g. `xhr_streaming` or `jsonp_polling` IIRC) and that would be interesting to know.

N3WWN, 2020-04-03:
I just refreshed and got the 'Loading failed' error.

Through dev tools, I see:
`Request URL: ws://192.168.42.253:5000/sockjs/384/4pewsvdd/websocket`

The IP and port are correct for my OctoPrint host.

Apparently the websocket is actually open because the data are currently streaming ():

[image]

N3WWN, 2020-04-03:
BTW, I wonder if this issue could possibly be related to another issue I'm having but haven't seen an issue for here in github:  At the end of prints, the web interface thinks the print has not yet completed.

Data (such as temp measurements) is still being sent from the printer to octoprint, which is proven by refreshing the web interface in the browser window, but the terminal and temp chart are both frozen.  Resetting the printer does nothing.

But, after a web browser reload/refresh, the current status of the printer is correctly reported and the temp chart displays all of the data, even the data received from the printer after the data freeze.

This isn't with every print, just a majority of them.

tanah, 2020-04-04:
Sorry for the delay in replying @foosel , I only see one GET request to the root but nothing else:

[image]

I reloaded a bunch of times in case something changes, as @N3WWN says that sometimes it comes through in his case. In mine I don't think I've ever seen it work past the update.

foosel, 2020-04-04:
You have a filter set, it currently only shows HTML, you'll need to set it to All/Todos.

tanah, 2020-04-04:
Ups, sorry. Yes, this is the one used in my case: `sockjs/133/lbabrwyg/websocket`

[image]

N3WWN, 2020-04-06:
> What I can do is increase the timeout on the `maintenance` branch and someone who can reproduce it can then check if that at least solves the symptom. I have no idea what could be taking so long in there, especially not under safe mode.

@foosel Just checking in to see if you’ve bumped the timeout in the maintenance branch.  If so, I can check it out and test.

Thanks!

foosel, 2020-04-07:
@N3WWN Now I have. And I've also made it configurable (via `config.yaml` only though). Setting `devel.sockJsTimeout` to 0 completely disables the timeout detection, a value >=1 sets the timeout to this many seconds. Default is 10s now (instead of 5s before). Pushed to `maintenance`, would be interesting to know if this solves it. If not it's something else/more that's going on.

N3WWN, 2020-04-08:
Thanks for making this change!

I have been testing it out since yesterday morning.  I am having some non-related issues with the maintenance branch, but upping the timeout did not initially help.  At 10s, it was failing to load the interface just as much as before.  I increased the timeout to 35 seconds, which seems to have helped a bit, but not entirely.  I just upped the timeout to 60 seconds and will see if that helps more today.

tanah, 2020-04-12:
I was going to install the maintenance brach but suddenly my installation seems to work just fine, even outside the safe mode. I have no idea what changed, do you want me to send any log file before switching to the maintenance brach just in case?

N3WWN, 2020-04-13:
I'm still having the same problem, even after increasing the timeout to 60 seconds.  I had just pulled the latest maintenance branch and updated my install.

BTW, I am also seeing these warnings in the logs:
```
2020-04-13 09:21:52,385 - tornado.access - WARNING - 404 GET /static/vendor/font-awesome-4.7.0/fonts/fontawesome-webfont.woff%3Fv%3D4.7.0 (192.168.88.177) 5.82ms
2020-04-13 09:21:57,497 - tornado.access - WARNING - 404 GET /static/vendor/font-awesome-4.7.0/fonts/fontawesome-webfont.woff2%3Fv%3D4.7.0 (192.168.88.177) 5.85ms
2020-04-13 09:21:58,793 - tornado.access - WARNING - 404 GET /static/vendor/font-awesome-3.2.1/fonts/fontawesome-webfont.woff%3Fv%3D3.2.1 (192.168.88.177) 5.50ms
```
and
```
2020-04-13 09:20:57,468 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_libs.js to /home/ralloway/.octoprint/generated/webassets/packed_libs.js.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:20:57,674 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_client.js to /home/ralloway/.octoprint/generated/webassets/packed_client.js.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:20:57,682 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_client.js to /home/ralloway/.octoprint/generated/webassets/packed_client.js.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:20:59,413 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_plugins.js to /home/ralloway/.octoprint/generated/webassets/packed_plugins.js.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:20:59,927 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_core.js to /home/ralloway/.octoprint/generated/webassets/packed_core.js.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:20:59,973 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_libs.css to /home/ralloway/.octoprint/generated/webassets/packed_libs.css.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:21:00,023 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_core.css to /home/ralloway/.octoprint/generated/webassets/packed_core.css.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
2020-04-13 09:21:00,067 - octoprint.server.util.webassets - ERROR - Error writing gzipped output of /home/ralloway/.octoprint/generated/webassets/packed_plugins.css to /home/ralloway/.octoprint/generated/webassets/packed_plugins.css.gz
Traceback (most recent call last):
  File "/home/ralloway/SeeMeCNC/octoprint/manual_install/OctoPrint_maintenance/lib/python2.7/site-packages/octoprint/server/util/webassets.py", line 127, in output
    gzipped_output.write(gzip.compress(data.encode("utf8"), compresslevel=9))
AttributeError: 'module' object has no attribute 'compress'
```

To get the page to load at all, I have to run `rm ~/.octoprint/generated/webassets/*.gz`.  

This is just in the maintenance branch.  I don't see this particular behavior (the failure to compress the webassets) with the 1.4.0 release.

Is there anything else I can do to assist in troubleshooting the web interface loading issue that results in the "Loading Failed" page?

foosel, 2020-04-13:
The gz issue should be fixed in the current version of `maintenance`, please update. There was an issue under Py2 with a PR that was merged and even though I even live streamed coding up the fix I forgot to push it.

Just out of curiousity, try to disable the timeout detection altogether (value of 0 as mentioned above) and see if you have any errors in the log and on the JS console then.

N3WWN, 2020-04-13:
Thanks for fixing the gz issue!  I was able to pull the updates for the `maintenance`  branch and did not have to remove the zero-byte .gz files this time.

Just to confirm, I'm setting the timeout in `~/.octoprint/config.yaml` like this:
```
devel:
    sockJsTimeout: 120
```

Is that correct?

I'm still getting the `Loading Failed` page even with sockJsTimeout set to 0.

Here's the JS console output:
[image]

...and the network info:

[image]

I let OctoPrint start fully and let it sit idle for about 6 seconds before I tried to load the page:

```
2020-04-13 13:43:05,364 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2020-04-13 13:43:05,558 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.42.253:5000/'}
2020-04-13 13:43:07,251 - octoprint.server.views - DEBUG - Forcing rendering of view _default
2020-04-13 13:43:07,455 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2020-04-13 13:43:07,598 - octoprint.server.util.flask - DEBUG - No cache entry or refreshing cache for / (key: ui:_default:http://192.168.42.253:5000/:en), calling wrapped function
2020-04-13 13:43:12,969 - octoprint.server.util.flask - DEBUG - Needed 5.37s to render / (key: ui:_default:http://192.168.42.253:5000/:en)
2020-04-13 13:43:12,971 - octoprint.server.preemptive_cache - INFO - ... done in 7.41s
```

When I tried to load the page, these are the logs generated:

```
2020-04-13 13:43:18,218 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.88.177
2020-04-13 13:43:18,278 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event PowerOn
2020-04-13 13:43:18,278 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event FirmwareData
2020-04-13 13:43:18,279 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event FileDeselected
2020-04-13 13:43:18,281 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event PostRollEnd
2020-04-13 13:43:18,281 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event FolderRemoved
2020-04-13 13:43:18,281 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event PowerOff
2020-04-13 13:43:18,280 - octoprint.events.fire - DEBUG - Firing event: ClientOpened (Payload: {u'remoteAddress': '192.168.88.177'})
2020-04-13 13:43:18,283 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-04-13 13:43:18,282 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event PrintDone
2020-04-13 13:43:18,283 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-04-13 13:43:18,285 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2020-04-13 13:43:18,284 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event UpdatedFiles
2020-04-13 13:43:18,287 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event MetadataStatisticsUpdated
2020-04-13 13:43:18,286 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-04-13 13:43:18,288 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-04-13 13:43:18,287 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event UserLoggedIn
2020-04-13 13:43:18,288 - octoprint.plugin - DEBUG - Calling on_event on SlicerSettingsParser
2020-04-13 13:43:18,289 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event plugin_softwareupdate_update_failed
2020-04-13 13:43:18,290 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event SlicingCancelled
2020-04-13 13:43:18,291 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event Disconnected
2020-04-13 13:43:18,291 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8c210490>> for event Setti [... cut ...]
2020-04-13 13:43:39,239 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8f80d1d0>> for event MovieDone
2020-04-13 13:43:39,240 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f5c8f80d1d0>> for event CaptureDone
```

I don't know how to verify this between the logs and the dev tools, but I believe the `Loading Failed` error showed up after the 3rd 404 error;  about 5 1/2 seconds later, the server logs make it appear that the page successfully loaded.

foosel, 2020-04-15:
I see some files being loaded there that aren't part of core OctoPrint and probably stem from a plugin (e.g. mixpanel.js). Not saying this is related, but it would still be interesting if you see the same issue in safe mode.

The 404s for the fonts on `maintenance` should be fixed by 2bed65d819c6c710ce1b56e0d18e03e87ab04422. 

Trying to figure out why setting the timeout to 0 still triggers the loading failed and I'm suspecting a caching issue introduced by more aggressive caching in `maintenance` through a PR, investigating.

N3WWN, 2020-04-15:
Yes... mispanel.js is used by octoprint_slicer.min.js, octoprint_slicer.min.js.map and slicer_tab.jinja2.

I updated to https://github.com/OctoPrint/OctoPrint/commit/2bed65d819c6c710ce1b56e0d18e03e87ab04422 and that did fix the 404s.  Thanks!  😃 

BTW, I installed 1.3.12 to a new virtualenv environment and and have had no issues with loading the interface on any of the browsers, apps or devices that I've tried, so it does seem to be something introduced with 1.4.0.

Not sure if this is useful info or not, but even when the 1.4.0 interface does load, it takes a LONG time for the files to populate, the printer to connect and temperatures to be displayed.  I'd guess somewhere around 45 seconds after the empty interface "framework" is displayed.

I'll check out safe mode in a little bit, if I get chance...  have to run at the moment.

N3WWN, 2020-04-15:
I've been able to do a bit of semi-exhaustive testing. 

With safe mode, the UI loads without delay.  The frame population still takes a while, but the empty UI loads right away and no 'Loading Failed' error.

So, I disabled every plugin and started re-enabling them slowly, restarting Octoprint to pick up the changes each time.

When I finally hit the 'Loading Failed' error, I disabled the last couple of plugins that I had just enabled... and no longer got the 'Loading Failed' error.  Then I started slowly re-enabling the rest of the plugins again, skipping over the couple that appeared to be the problem only to find that enabling additional plugins at all would trigger the 'Loading Failed' error.  

It doesn't appear to be particular plugins, but perhaps the combined total memory used by the plugins or some other combination of enabling multiple plugins that causes the problem.

Is there a limit to how many plugins should be used?  I didn't think I was going overboard with the following list:

* Detailed Progress
* DisplayZ
* EEPROM Repetier Editor
* Exclude Region
* Firmware Updater
* GCode Editor
* Navbar Temperature
* Preheat
* Simple Emergency Stop
* Slic3r
* Slicer
* SlicerSettingsParser
* SlicerSettingsTab
* Status Line
* STL Viewer
* Tempsgraph
* Terminal Commands Extended

Does any of this help?

foosel, 2020-04-20:
My money is less on the combined memory usage and rather on the amount of requests triggered by the other plugins delaying things so much that loading fails. However that's fairly unusual in itself - there are people out there with way more plugins than you and I haven't heard from any instances of this issue than the two reported cases on this ticket so far. I would expect this issue to be way more widespread if the problem is a certain number of plugins. 

Just a hunch, can you please disable the Slicer plugin? I want to make sure it's not this mixpanel stuff interfering. And after you've done that another screenshot of the Network tab would be great.

To give you some background, the Loading Failed message is simply a connect timeout trigger that gets called if the connection to the push socket couldn't be established (incl. a handshake from the server) within the now configurable timeout. This message was introduced in 1.4.0, so I'm not surprised you didn't see it in 1.3.12 (or in a fresh venv without plugins). That you see it indicates delays in the page load. 

What I cannot reproduce at all at the moment is setting the timeout to 0 *not* disabling the Loading Failed message altogether. There's no caching issue as I originally assumed, I just toggled things back and forth and the Loading Failed detection got disabled correctly. Just to verify, you did set `devel.sockJsConnectTimeout` to 0 in `config.yaml`:

``` yaml
devel:
    sockJsConnectTimeout: 0
```

saved, then restarted the server?

dshokouhi, 2020-04-24:
I am seeing the same issue intermittently on my end.  Octoprint actually loads sometimes for me.   I find sometimes I have to refresh the tab a few times for it load and other times it loads very quickly.

Here is my list of plugins:
Action Command Prompt
Bed Visualizer
BetterHeaterTimeout
Cancel Objects
Cost Plugin
Dashboard
DiscordRemote
Fan Speed Control
Filament Manager
Floating Navbar
GcodeEditor
M73 Progress
Navbar Temperature Plugin
Octolapse
Print Track
PrintTimeGenius Plugin
Printer Stats
PrinterAlerts
Tab Order
Themeify
TouchUI
Yamlpatcher

Console error log: https://pastebin.com/CVYkPfzi
Octoprint log: https://pastebin.com/i88rJ5HJ

I have a friend who has never seen this issue and does not have the amount of plugins that I do so I suspect it may be related to some of them or one of them :)

N3WWN, 2020-04-24:
@foosel ...

> Just a hunch, can you please disable the Slicer plugin? I want to make sure it's not this mixpanel stuff interfering. And after you've done that another screenshot of the Network tab would be great.

As requested ;) 

[image]

> To give you some background, the Loading Failed message is simply a connect timeout trigger that gets called if the connection to the push socket couldn't be established (incl. a handshake from the [...]

I can confirm delays in items loading, and I think that's apparent in the screenshot of the Network tab, too. 😉 

> What I cannot reproduce at all at the moment is setting the timeout to 0 _not_ disabling the Loading Failed message altogether. There's no caching issue as I originally assumed, I just toggled [...]

Aha!  That may be why disabling the timeout wasn't working!

In https://github.com/OctoPrint/OctoPrint/issues/3502#issuecomment-610295139 , you said the configuration item was `devel.sockJsTimeout` .  

The UI loads even after a 35s delay using `devel.sockJsConnectTimeout` set to 0!

foosel, 2020-05-15:
I still haven't been able to reproduce, however I've slightly change the order in which things are getting triggered on initial bootup of the frontend. It would be interesting if the current `maintenance` branch makes the issue go away even with `devel.sockJsConnectTimeout` to something like 10s. 

Also apologies for the mix up in names there 😱

jbbandos, 2020-05-24:
I'm following this, as I have the same issue. In my case, there seems to be a particularity - the first chrome browser I open a connection to octoprint works most of the time, and then starts slowing down and not answering clicks immediately (might answer after a few seconds delay). 
Brave and waterfox don't seem to have any problem - as first or second browser, even both in parallel with chrome - which is strange as chrome and brave both have the same engine and I have the same extensions on both. Opening chrome as the second or third browser connecting to octoprint just fails with the same loading failed error as others have documented.

gege2b, 2020-05-24:
> My money is less on the combined memory usage and rather on the amount of requests triggered by the other plugins delaying things so much that loading fails. However that's fairly unusual in itself [...]

I get this error randomly (but mainly when I launch firefox and switch to the octoprint's pinned tab)

In my case trough, I put this on my rpi being a bit too far from the wifi AP and didn't pay much attention about it, usually, a simple refresh is enough. 

If this can help, next time, I'll do a proper report (given that this may simply be caused by an slightly unstable connection)

foosel, 2020-08-04:
Can y'all please test if this still occurs against the just release 1.4.1? If I don't hear anything back about this within the next seven days I'm going to assume the changes I made mentioned earlier solved it.

N3WWN, 2020-08-05:
@foosel I tested OctoPrint 1.4.1 today and the issue seems to have completely disappeared when I use a desktop browser. 😁  At least I was unable to replicate the issue with a desktop browser (Chrome).

But, when I use OctoShell on my iPad Pro 10.5, I still get the "loading failed" error. 😢 

BTW, to test this, I removed the following lines from my configuration:

```
devel:
    sockJsConnectTimeout: 0
    sockJsTimeout: 0
```

foosel, 2020-08-06:
I mean, the problem might still present if loading in fact DOES take too long for the default timeouts.

Does it help to increase them, and if so to what values?

N3WWN, 2020-08-06:
@foosel Good point ;)  I just tested by increasing `sockJsConnectTimeout` in steps from 1 until I found that at 15s, my iPad Pro 10.5 would load the interface consistently.

I realize that 10s is the default, but I wanted to verify that I had the timeout correctly configured in the config file.

When set below 5s, the timeout was appropriately short and increased as I bumped up the timeout. 

I'm not sure if 15s is a tipping point or not, though.  With more load on either the server or client, or more complexity/plugins/etc, it may not load (or load consistently), but for now, 15s is working well.

foosel, 2020-08-06:
I'll look into bumping it to something higher then. The problem with finding such threshold values is always that as you said they can be highly reliant on the user's environment, so I'm open to some (sensible) increases here. Will set it to 20s.

tanahy, 2020-08-06:
After updating to 1.4.2 I've been restarting one of my boards and 
reloading the interface a good amount of times and it seems be working 
just fine except for a couple of times. However I have tried with a new 
power supply that just arrived today and it seems to load faster or at 
least it seems to be the case and doesn't result in the expected problem.

As such, I think my problem was in part (or maybe fully) due to the 
power supply. Thanks for the update and I think you won't need more 
feedback from me.

foosel, 2020-08-11:
Considering this mostly solved already and will close when 1.5.0 ships with the timeout set to 30s by default.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel