OctoPrint/OctoPrint #3505: [Request] Add unselect to /api/files in REST API

OutsourcedGuru, 2020-04-02:
**Is your feature request related to a problem? Please describe.**
It would be great if programs could unselect a file so that file queuing mechanisms, for example, might be written. There are times when you'd like to simply unselect the currently-selected file if you made a mistake, for example.

**Describe the solution you'd like**
In the POST to `/api/files/local/filename` endpoint of the API, the `command` could be `unselect` rather than `select`. I couldn't imagine other ways of indicating this (that might be more appropriate) other than possibly using a keyword like `/api/files/local/none`, perhaps with the standard `select` as the `command`...?

It feels odd having to indicate the filename so that it could be unselected. In some ways, I think I prefer the `/none` version of this.

**Describe alternatives you've considered**
I'm currently using the internal interface of printer.unselect_file(). But this requires you to be running within the space of a plugin.

**Additional context**
N/A

NotExpectedYet, 2020-04-06:
Would totally love this myself, it's been a requested feature for OctoFarm and I'm currently trying to work away around it with the current API implementation.

cp2004, 2021-10-11:
This was implemented in 1.7.0 which has been released 🎉

closed 2021-10-11 by cp2004