OctoPrint/OctoPrint #3508: 1.4.0 rebooting restores a timelapse setting I have shut off (and saved)

iandouglas, 2020-04-06:
#### What were you doing?

I needed to repurpose the webcam that I had on a Pi 4, running 1.4.0. Without a webcam installed, obviously I was getting errors about the webcam not being present, I went into the Timelapse tab, turned timelapse capture to "Off", saved my changes, and carried on with printing.

On occasion, I've had to reboot my Pi 4 because my Prusa Mini has needed to be powered off or reset and the old USB port is no longer detected properly. Upon reboot, then found that the timelapse setting was turned back ON and the UI complaining about the inability to find a webcam to take timelapse photos, and also an error upon print completion that a timelapse video could not be made.

#### What did you expect to happen?

I would expect that turning off the timelapse setting, and saving my changes, would persist that setting even after a reboot of the Raspberry Pi.

#### What happened instead?

The timelapse setting was re-enabled through no user manipulation whatsoever. (I'm the only one with access to these Pi's)

#### Did the same happen when running OctoPrint in safe mode?

I haven't tried this, I've been printing PPE non-stop and this was more an annoyance than an urgent "stop everything and try to debug this further" ordeal.

#### Version of OctoPrint

1.4.0 (Stable)

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

Prusa MK3S and Prusa Mini, both with latest stable production firmware.

#### Browser and version of browser, operating system running browser

Chrome Version 80.0.3987.163, Mac OS Mojave

#### Link to octoprint.log

Current log is too big for pastebin (4MB). Unsure what I can trim out that would/wouldn't be helpful.

#### Link to contents of terminal tab or serial.log

I'm actively printing PPE, so cannot get anything useful from Terminal, but I'll try to reproduce this and update my bug report in a few hours when this mask is done.

#### Link to contents of Javascript console in the browser

Same as above, I'll try to reproduce this later this evening.

#### Screenshot(s)/video(s) showing the problem:

n/a

[x] I have read the FAQ.

iandouglas, 2020-04-09:
Update, this also happened this morning when I updated some plugins and OctoPrint merely restarted (not a full Raspberry Pi reboot). This has happened on 3 separate OctoPrint installations, all running 1.4.0.

iandouglas, 2020-05-13:
Sure, but I need to wait on a new webcam to arrive, I had to repurpose two or three for remote teaching.

iandouglas, 2020-05-14:
I installed a handful of plugins that I normally use, toggled timelape mode "on" in "timed" mode, restarted, and it was back off. Restarted in safe mode, toggled it "on" in "timed" mode, restarted in safe mode and it was off again, rinse and repeat.

[iandouglas_octoprint.log](https://github.com/OctoPrint/OctoPrint/files/4630143/iandouglas_octoprint.log)

kantlivelong, 2020-05-14:
@iandouglas Thanks for the logs. Out of curiosity did you make sure to check "Save as default" and then click save?

iandouglas, 2020-05-14:
In safe mode, "Save by default" was off, I enabled it and restarted in safe mode and it remembered to come back as "on" and "timed" but the "save by default" flag is off now. Restarting again in safe mode, and it's still in "timed" mode. Restarting in "regular" mode (not safe mode), and I see the setting is also still set to "timed" mode.

This was a brand new Pi setup as of this morning since my other systems are still printing PPE. I *have* set the "default" flag in the past, and yet they weren't remembering the setting. I wonder if there was something quirky in an upgrade path? (I've been using OctoPrint for 2? years now and generally upgrade as soon as a new stable release comes out) I'll double check the other Pi systems when some PPE is done and see if it's still happening there. Just odd that once I pulled the cameras off the other Pi's that this would start happening.

kantlivelong, 2020-05-14:
Just to be sure. It is working now but the "Set as default" checkbox isn't being remembered? If so that is expected.

iandouglas, 2020-05-18:
Cannot reproduce now.

closed 2020-05-18 by iandouglas