OctoPrint/OctoPrint #3512: [1.4.0] Cannot generate new API key

paukstelis, 2020-04-10:
#### What were you doing?
Trying to change API key

Enter settings, API, click 'Generate new API key' button

#### What did you expect to happen?
API key would change
#### What happened instead?
Nothing.
#### Did the same happen when running OctoPrint in safe mode?
Didn't try.

#### Version of OctoPrint
1.4.0

#### Operating System running OctoPrint
Linux

#### Printer model & used firmware incl. version
Not applicable

#### Browser and version of browser, operating system running browser
Brave/Firefox

#### Link to octoprint.log
Nothing logged.

#### Link to contents of terminal tab or serial.log
Nothing logged.

#### Link to contents of Javascript console in the browser
Nothing shows up.

#### Screenshot(s)/video(s) showing the problem:
Nothing to show

I have read the FAQ.

foosel, 2020-04-10:
Try safe mode and please share the logs regardless, there's valuable information about your runtime environment in there even if no error is being logged.

paukstelis, 2020-04-10:
Went through and tested on a new installation and it worked fine, so it is probably something with this specific install. Will reinstall when I can to confirm that it is not something system specific.

foosel, 2020-06-12:
@paukstelis any update on this?

paukstelis, 2020-06-12:
Sorry. This had to be some remnant from some update process gone wrong along the way. Fresh install did not have the issue. Closing.

closed 2020-06-12 by paukstelis