OctoPrint/OctoPrint #3513: Py3: octoprint.util.to_unicode -> UnicodeDecodeError

OllisGit, 2020-04-10:
#### What were you doing?

Processing gcode-lines via LineProcessorStream in a Python 3 environment.
For that reason I convert the provided line-variable from bytes to str.

In most cases it works, but not if the the gcode includes "latin/french-letters", see Issue https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/126 

Python-Script to reproduce the error:
```
import octoprint.util
# gcode_line_as_str = "M117 Priming Filamentâ{¦"
gcode_line_as_bytes = b'M117 Priming Filament\xe2{\xa6\n'
print (gcode_line_as_bytes)

unicode_line = octoprint.util.to_unicode(gcode_line_as_bytes)
# --> BOOOM: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 21: invalid continuation byte
```
My current solution at the moment is: Instead using the util-function I use ```gcode_line_as_bytes.decode('ISO-8859-1')```
Do not use ```utf8```!!!! 

#### What did you expect to happen?
just convert the bytes to str

#### What happened instead?

```UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 21: invalid continuation byte```

#### Did the same happen when running OctoPrint in safe mode?
no, because my plugin is then disabled

#### Version of OctoPrint
1.4.0

#### Operating System running OctoPrint
MacOS

#### Printer model & used firmware incl. version
not relevant

#### Browser and version of browser, operating system running browser
not relevant

#### Link to octoprint.log
not relevant

#### Link to contents of terminal tab or serial.log
not relevant

#### Link to contents of Javascript console in the browser
not relevant

#### Screenshot(s)/video(s) showing the problem:
not relevant

I have read the FAQ.

foosel, 2020-04-10:
Take a look at the available parameters on that utility methods. It supports encoding and errors behaviour just as encode. If you give it a iso-8859-1 byte string and tell it to turn that into a utf8 unicode string it might run into these issues. errors="replace" is probably your friend here.

closed 2020-04-10 by foosel

OllisGit, 2020-04-11:
Thx, for the fast response. Now I use the errors-behaviour.