OctoPrint/OctoPrint #3520: Add PrintKilled event

skorokithakis, 2020-04-13:
In some rare cases, printers might report a catastrophic problem (thermal runaway, other failures). In those cases, I might want something different than a failed print. For example, currently I send a message to my phone on a failed print, but on a "printer killed" event I want to send a message to the printer relay to shut it off immediately.

This isn't very high priority, since it's not very likely that I'll catch anything that a circuit breaker won't catch, but it might be useful to add.

One note is that, when the PrintKilled event is triggered, the PrintFailed event should probably also trigger.

foosel, 2020-04-14:
In those cases the [`Error` event](https://docs.octoprint.org/en/master/events/index.html#printer-communication) should already be triggered and be available to react to. It's raised for all non recoverable errors reported by the firmware that lead to print cancellation and/or disconnect:

[image]

Since errors are not standardized in any way, it's exactly how your `PrintKilled` event would need to be implemented.

closed 2020-04-14 by foosel

skorokithakis, 2020-04-14:
Ah, I didn't notice that, sorry. Thanks for the help.