OctoPrint/OctoPrint #3521: Rendering Timelapse Failed Error

leitenky, 2020-04-14:
#### What were you doing?

1. Go to settings > Timelapse > advanced
2. Click on x264 redering in place of mpeg
3. Perform a print
4. Error occurs: Rendering Timelapse Failed, if you expand the error there is no details

#### What did you expect to happen?
The video to render. I have two pis running octoprint (a 3b and a 3b+) both have this issue

#### What happened instead?
Received an error "Rendering Timelapse Failed"

#### Did the same happen when running OctoPrint in safe mode?

Safe Mode disables the GPX plugin which is required for my printer to run

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Wanhao Duplicator D4S

#### Browser and version of browser, operating system running browser

 Chrome (windows) and Touchui (w/ Chromium)

#### Link to octoprint.log

2020-04-14 00:17:11,872 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2020-04-14 00:17:11,933 - octoprint.util.commandline - ERROR - Error while trying to run command /usr/bin/avconv -framerate 20 -i "/home/pi/.octoprint/timelapse/tmp/Basic_Bias_Tape_Folder_Set-Fast_20200413224325-%d.jpg" -vcodec libx264 -threads 1 -r 25 -y -b 10000k -f mp4 -vf '[in] format=yuv420p [postprocessed]; movie=/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/static/img/watermark.png [wm]; [postprocessed][wm] overlay=10:main_h-overlay_h-10 [out]' "/home/pi/.octoprint/timelapse/.Basic_Bias_Tape_Folder_Set-Fast_20200413224325.mp4"
2020-04-14 00:17:11,934 - octoprint.timelapse - WARNING - Could not render movie, got return code None: []

#### Link to contents of terminal tab or serial.log

NA

#### Link to contents of Javascript console in the browser

NA

#### Screenshot(s)/video(s) showing the problem:

 
[image]

I have read the FAQ.

foosel, 2020-04-14:
> It appears that path to FFMEPG is incorrect?

Did you update the OctoPi image and restored from a backup? Try if `/usr/bin/ffmpeg` works.

leitenky, 2020-04-14:
No I didn't restore from a backup. The strange thing is the issue started after updating to the latest octoprint (1.4.0), I have not touched timelapse settings in over a year.

The other strange thing was both my octoprint servers have the same issue (set to AVCONV instead of ffmpeg) even though they are both set up for completely different uses (3D printer and CNC)
Perhaps an issue with the upgrade? 

Long story short switching the setting back to /usr/bin/ffmpeg worked

foosel, 2020-04-15:
avconv used to be what you needed in Raspbian up until about a year or so ago when they switched ffmpeg back to being called ffmpeg. Hence the question whether you might have restored a backup on a newer version of OctoPi. OctoPrint didn't touch this setting, it doesn't even come preconfigured on stock OctoPrint, that's all customized in OctoPi during build of the image. What could be is that some third party plugin modified it, or maybe you still had an older avutils package on your install, did an apt-get upgrade and that replaced it with the new ffmpeg only version and thus stuff stopped working. What I can tell you is that it really cannot have been the upgrade to 1.4.0 because that didn't touch that setting (nor did any updates before it).

Closing since setting the correct path works, no path is preconfigured in OctoPrint so nothing to fix there and last I checked it had been updated in OctoPi.

closed 2020-04-15 by foosel