OctoPrint/OctoPrint #3522: [Request] Direct SD-Card upload

Sineos, 2020-04-17:
**Is your feature request related to a problem? Please describe.**
I prefer printing over the SD-Card as quite some model do not well print over serial. As well known, transferring a gcode-file to the SD-Card is painfully slow. My current workflow is as follows:

**One time setup:**

1. Install usbmount
2. Install Samba server
3. Configure Samba to serve the SD-Card mount point 

**Printing workflow:**
1. M22 ; Release the SD-Card
2. Transfer the gcode file from my remote computer to the SD Card via SMB
3. M21 ; Init the SD Card
4. Refresh SD Card content in OctoPrint
5. Print

**Describe the solution you'd like**
I would like to have OctoPrint perform steps 1 to 5 transparently as new upload method

**Additional context**
This might also be a way to solve the long standing 8.3 filename discussion. When the SD-Card is mounted to the operating system, the proper filenames can be retrieved via a single FS call.

foosel, 2020-04-17:
That is too specific to a very special kind of setup that's not generally available to the majority of the user base (printer's SD that *can* be mounted and *is* mounted to some location on the file system) and thus ill suited for inclusion in core.

What I'd rather like to look into is abstracting the SD file access into a proper storage module and open addition (and possibly replacement) of storage modules via the plugin API, and copy/move between storages in general (with capability modelling that marks whether a storage supports that or not). 

What you want to do should actually already be possible to do with a somewhat hackish approach from a plugin though, so I'm marking this as a plugin idea.

DanielJoyce, 2020-05-07:
I assume you have your computer or pi connected to the printer via USB, and also supports mounting the SD card in the printer over the usb to the pi host? Is that correct? What model of printer?

Sineos, 2020-05-08:
It is a SKR 1.3 board on a Creality Ender 5.

SKR board is connected via USB to the RPi running Octoprint. When I want to transfer data to the SD, I'll release the SD card from the firmware. Then it will get connected to the RPi as USB mass storage device. This mass storage device I mount as a Samba drive and it can be easily accessed from anywhere in the network.

With "usbmount" on the RPi this works pretty much automatically. The only manual thing to do, are the steps in the first post "Printing Workflow".
I get around 400 kB/s transfer speed to the SD card over the network. Actually I like the solution.

moham96, 2020-12-24:
I also have a similar setup, in my case, I prefer to print from sd card because power loss recovery only works if you print from sd card and I don't have easy access to remove and reinstall the sd card each time to put the g-code file.
Octoprint uses the `m28` which is **painfully** slow.
what I would like octoprint to do is when I upload a file to sd card using the web UI it should run `M22` the waits until the sd card is mounted and copy the g-code to it, then run `M21` to init the sd card so I can see the file and print.

Regards

cp2004, 2020-12-24:
Could still be done with a plugin - not sure what hardware is needed for the USB mount of the SD card, I don't know if mine does. If this is not widely supported by most hardware/firmware then it is best suited for a plugin rather than core OP.

If you want a solution *now*, then maybe look into the binary file transfer plugin, which should be faster than the line-by-line streaming.

arekm, 2022-08-04:
Try https://plugins.octoprint.org/plugins/sdwire/ and set sd-mux-ctrl binary path in settings to /usr/bin/true (+ proper uuid).

jneilliii, 2022-08-04:
Be warned, there was a user on discord that broke their connection to their printer messing with sd-mux-ctrl if done improperly.

arekm, 2022-08-04:
sd-mux-ctrl binary is not needed in this setup. From earlier comments it looks like only M21/M22 are needed to switch sd card between printer and usb-storage on SKR boards.