OctoPrint/OctoPrint #3524: Multiple layer shift when printing from OctoPrint

radimsejk, 2020-04-18:
#### What were you doing?
It happened a lot of times but lastly I have been printing a flover pot overnight. After 3 hours of print it started layer shifting every hour or so. When I print the exact same gcode from sd card it does print well.

#### What did you expect to happen?
I expect not to have layer shifts on my prints.

#### What happened instead?
It shifted layers several times and when I hit pause after 9 hours of print to inspect inside of the flower pot it crashed printer head to the flower pot so I had to turn off the printer.

#### Did the same happen when running OctoPrint in safe mode?
It is happening only on large prints, so I do not want to waste filament and time on it :(

#### Version of OctoPrint
1.4.0

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
Prusa MK3S with latest 3.8.1

#### Browser and version of browser, operating system running browser
Chrome 80.0.3987.163 on Windows 10

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log
Do not have serial.log nor the terminal tab active now.

#### Link to contents of Javascript console in the browser
Do not have that.

#### Screenshot(s)/video(s) showing the problem:
first layer shift https://www.twitch.tv/videos/595369901
crash when paused on the end https://www.twitch.tv/videos/595372747
whole print until crash https://www.twitch.tv/videos/595020517

FixxCZ, 2020-04-19:
I have the same issue on the same version and same printer. Attaching serial and octoprint log. I started the serial one after first layer shift that was somewhere around 15:00. Second shift happened with serial log enabled somewhere around 16:15.
I'm have to reprint it tomorrow and I will try it in safe mode.
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint/files/4500273/octoprint.1.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/4500274/serial.log)

radimsejk, 2020-04-20:
I am finally restocked on PLA so I will try to print something big in safe mode too and attach logs. I hope this will be solved because without Octoprint it is only half of my printer :)

foosel, 2020-04-20:
@radimsejk Please share logs, I cannot analyse anything like this. Also please respond to the question by @FixxCZ because I'm seconding that. 

I have never seen a layer shift caused by OctoPrint, it doesn't have code that would even make this possible, so my money is on either something like a third party plugin causing trouble with some injected GCODE or the steppers are overheating or something.

radimsejk, 2020-04-20:
@foosel and @FixxCZ Yes I do have the mentioned plugin. I will try to disable it and print without it. Thank you for your help :)

foosel, 2020-04-20:
So here's what I did... I took the `serial.log` from @FixxCZ, stripped the `Recv` lines and stripped the remaining lines down to the commands themselves that went over the line without line numbers or checksums. Saved that as `gcode` and loaded it into OctoPrint's GCODE viewer.

[image]

I cannot see a layer shift in these lines. So whatever is happening, from the looks of it has nothing to do with the GCODE that is being sent, at least not in this example. If it was the movements being in any way manipulated by OctoPrint to cause a layer shift, that would be visible here in form of a shift in the viewer. 

The only thing I see a number of `M402` that surprised me in their number. I don't know how the planner reacts to acceleration reconfiguration during an active stepper hold. 

---

**Questions & tasks for everyone running into this issue:**

  * does it happen in safe mode as well? if so, what plugins do you have installed?
  * does it only happen on specific jobs?
  * does it also happen when printing from your printer's SD card?
  * when did it start happening?
  * what axes are effected? Only one? X, Y, Z, maybe even E? Does that change?
  * share the following:
    * `octoprint.log`
    * a `serial.log` from a reproduction (*you need to enable `serial.log` first!*)
    * a gcode file that shifted
  * rule out mechanical & environmental reasons for your shifts (slipping pullies, loose belts, overheating steppers, undervoltage, ...)

foosel, 2020-04-20:
**Questions & tasks for everyone running into this issue:**

  * rule out mechanical & environmental reasons for your shifts (slipping pullies, loose belts, overheating steppers, undervoltage, ...)
  * share the following:
    * printer model
    * printer firmware version
    * OctoPrint version  
    * `octoprint.log`
    * a `serial.log` from a reproduction (*you need to enable `serial.log` first!*)
    * a gcode file that shifted
  * does it happen in safe mode as well? if so, what plugins do you have installed?
  * does it only happen on specific jobs?
  * does it also happen when printing from your printer's SD card?
  * when did it start happening?
  * what axes are effected? Only one? X, Y, Z, maybe even E? Does that change?

Test & share **all** of the above, I need **all** of this to be able to analyse this issue, none of this is optional, all of it is required. Please cooperate and do your part by providing me with the data I need.

radimsejk, 2020-04-20:
- All belts and pulleys are snug, power is OK and all affected axis has warm not overheating steppers.
- I have printed two designs but smaller ones with no issue in safe mode. Will try something bigger tonight.
- No, it has happened three times to me on different jobs.
- I have reprinted the failed jobs from SD card with no issue.
- I am using octoprint for just few days
- It affects X and Y on my prints. Sometimes only one axis, sometimes both.
- Will share logs after some failed print. I have no logs from the previous failed prints.

FixxCZ, 2020-04-20:
I had run the same 6 hours print today in safe mode and it was perfect. No layer shifts at all. This print had three shift yesterday. However the plugin that I suspect (Display Layer Progress) was disabled yesterday (not uninstalled, just disabled). Could it be causing the issues or I should look for another plugin that was causing it. I had number of them installed as I was hoping that Display layer Progress was the cause and was trying new plugins.

radimsejk, 2020-04-21:
I have run 10 hour print last night with DisplayLayerProgress disabled from plugin manger and it completed flawlessly. I have now started 6 hour print and will print for 12 hours later on so we will see if that helped for 100%. So I will not post the logs because print was sucessfull.

radimsejk, 2020-04-23:
I disabled the DLP plugin and had multiple layer shifts in latest print right now. With no visible problem on video.

Video: https://www.twitch.tv/videos/600091683
Logs and Gcode downloaded from Octoprint: [Link to Dropbox](https://www.dropbox.com/sh/gsz19pm5bovjtec/AABp3QWoTRt2TpBG5C94Beyfa?dl=0)

Prusa MK3S 3.8.1
DisplayLayerProgress v1.20.0
Octoprint 1.4.0
OctoPi 0.17.0

FixxCZ, 2020-04-23:
What other plugins do you have? I had the issue as well with DLP disabled. But I blame Octolapse from that one.

foosel, 2020-04-23:
@radimsejk I did the same thing with your log that I did with the one that @FixxCZ shared earlier: extracted the commands that OctoPrint sent to your printer, saved as `gcode` file and then loaded that into OctoPrint's own GCODE viewer:

[image]

There's no coordinate shift visible in the commands, what OctoPrint sent to your printer should have produced a flawless printing result. So whatever is causing the actual layer shift is something that's either in your firmware or mechanical.

As I understand it, your layer shift still happened while OctoPrint was not running in safe mode. Please try to reproduce your issue in safe mode, I need to know if it happens there as well or not. You have quite a number of third party plugins installed that hook into the serial connection.

@FixxCZ What printer do you have? The DLP issues seem to happen primarily (though not exclusively) with Prusa printers from what I read on that ticket, and I'm starting to suspect a connection between their firmware and this issue here, possibly routed in low running movement planning buffers or similar.

radimsejk, 2020-04-23:
I have:
Dashboard (1.11.4) (that is why I have DLP)
Exclude Region (0.2.0)
Fullscreen Plugin (0.0.4)
GCODE System Commands (0.1.1)
LayerDisplay (0.4.3)
MQTT (0.8.6)
Octolapse (0.3.4)
Printer Stats (2.0.2)
Prusa ETA override Plugin (0.1.1)
PrusaSlicer Thumbnails (0.0.3)
Pushover (0.3.3)
TouchUI (0.3.14)

radimsejk, 2020-04-23:
@foosel Ok, I will try to print with safe mode only from now on. Let's see. BTW I am printing now the same gcode as before and there is still no layer shift and it is only a few minutes to the end.

FixxCZ, 2020-04-23:
@foosel I have a week old i3 MK3S assembled by Prusa with FW 3.9.0 RC1. I'm setting up the printer as shared printer within our company and expected various issues but not Octoprint issues. However safe mode worked for me but don't want to use vanila Octoprint. Currently having DLP V1.20.1dev wihtout a problem, but had a very few prints so far to tell if that worked.

foosel, 2020-04-23:
So both of you here run Prusa FW. 

Do both of you have crash detection and/or the filament sensor activated?

The only layer shift I ever saw on my MK3, completely at random and on a more or less vanilla, was with crash detection enabled (years ago at this point, disabled it and never reenabled it again)

radimsejk, 2020-04-23:
@foosel Yes, I have crash detection enabled. But when the layer shift occurs the printer does not say it detected any crash. It just shifts the layer and continues.

foosel, 2020-04-23:
Here's a suggestion. Try printing in safe mode for a while. If that fixes things, we'll take a closer look at your plugins.

If not or upon return to normal mode, reproduce the issue, switch off crash detection and try to reproduce again.

It's just a hunch to be honest, based on my own anecdotal evidence, but imho it's worth a try since the whole issue doesn't make sense to begin with (prints stuttering would, but a layer shift when the coordinates that are being sent are correct shouldn't be possible to induce by serial commands and that's the only thing OctoPrint can send to the printer, so taking out stuff that messes with the stepper control routines inside the firmware might be worth a shot).

Please understand that I'm not saying there are no layer shifts, I'm saying that that they are happening *should* be impossible. The only way that streaming a GCODE file to the printer through serial should be able to mess up a print is by streaming too slow and hence causing stuttering and blobbing. There's simply no perceivable way how to make the printer intentionally or accidentally perform a layer shift via serial. So if that is happening, there's at least a partial fault for this inside the firmware (mechanical issues being ruled out that is).

FixxCZ, 2020-04-23:
I have crash detection enabled as well. I won't be able to run in safe mode as the printer will be deployed to production next week and needs to have few plugins enabled. But I can give a try to the crash detection disabling.

radimsejk, 2020-04-28:
@foosel  So i have been printing in safe mode for 4 days with some small prints and two 16 hour prints. Everything is flawless. I will try next to print in normal mode with every plugin disabled and with every three large prints I will enable one more to see which plugin will be the one. I will start with the DLP. 

Is this approach right or do you have any other idea? And thank you for your interest in this.

foosel, 2020-04-28:
Sounds like a solid troubleshooting approach to me.

FixxCZ, 2020-04-28:
I did 11 hours print yesterday with crash detection off and the newest version of the DLP plugin and all went perfect. There was even a moment when I loaded a Gcode for next print and RPI start analyzing it which put a significant load to RPI. It slowed down print for a 20 second but did not shift the layers.

foosel, 2020-04-28:
@FixxCZ Just saw your edit up there...

> What it will does when it crash with detection being off? It will simply try to push as hard as it can? Or stop and could not resume the print?

It will continue to try to print. As far as I understand its implementation, crash detection works the same way as the homing detection and will try to rehome if it encounters some kind of interference on its path. And that might be where things go south. 

It's certainly interesting that your number of detected crashs seem to correspond to your number of encountered shifts.

radimsejk, 2020-05-05:
So I have been printing a lot lately and with enabling plugins one after each other I had not any layer shift since (maybe one of latest updates solved it). But I had another weired bug today. I have been printing four instances of a discs as you see on the photo. When it finished the infill first full layer was next. On the first disc it went good. On second just a half of the disc was layered and than it finished next two discs without problem. I looked at the GCode Viewer and Octoprint was thinking that it finished whole layer. The photo is taken when the second full layer was on.
[image]

radimsejk, 2020-05-11:
I had a layer shift just right now. With DLP disabled. @foosel do you want some logs, gcodes or anything after it finishes?

Prusa i3MK3S 3.8.1
Raspberry Pi 4B with OctoPrint 1.4.0 running on OctoPi 0.17.0
[image]

foosel, 2020-05-12:
Was linear advance enabled? Was crash detection enabled? Was the filament sensor enabled?

I stand by what I've already said on the DLP ticket, whatever that is, it has a firmware component as well. The only thing I can see OctoPrint cause is stuttering. Data not being pumped fast enough to the printer should *never* cause it to shift coordinates in an axis, so something else is going on here. The above features are possible candidates.

Current guess is that these shifts are related to buffer underruns caused by data not being pumped fast enough. What is actually causing the shifts on firmware side then is still unclear, but in any case a firmware issue. LA would be a candidate that's available across printer brands and which has caused issues due to calculation overhead in the past. Crash detection also has caused trouble in the past.

I suggest to disable all that stuff and see if the issues persist.

radimsejk, 2020-05-12:
Everything was enabled. I can disable the filament sensor and crash detection however I do not know how to disable linear advance. Is it that simple as deleting this line in my slicer?
[image]

foosel, 2020-05-12:
I think that should do it.

radimsejk, 2020-05-12:
> I think that should do it.

OK, I will try it from now on. Let's see if it helps.

radimsejk, 2020-05-18:
OK, I have turned all the features off. Another layer shift happened. [image]
I than continued with turning off my plugins one by one and OctoLapse was next. Since then I had all prints successfull. I am not saying that it is 100%, I need to print more with this plugin disabled and other plugins enabled, but I hope I am on the right way to find the reason of this problem.
Maybe I have some bad settings in this plugin. We will see.

foosel, 2020-06-18:
Have you made any new discoveries here?

radimsejk, 2020-06-18:
I sadly have not. I have now my printer offline due to extruder heater failure.

mrseeker, 2020-06-21:
I am adding myself here to the list of people that are experiencing "layer shifts" in the print.

Octoprint 1.4.0, running an Ender-3, Marlin firmware 2.0.5.3. Some prints run fine, some prints tend to go "layer shifting" at random parts of the print, but they are more noticeable when I print with a thicker layer height. I am unsure what is causing the issue, but I am pretty sure I cannot reproduce it on the "same" height. If it was a print issue, I would have seen it on various prints, however the layer shift is irradic and not reproducable. It is also not just "a single axis", both X and Y-axis were affected by the layer shift.
I am not ruling out anything at this point, but it might be possible that the feed rate might trigger a floating point arithmetic problem?

mrseeker, 2020-06-23:
Mine might be related to a known Marlin bug: https://github.com/MarlinFirmware/Marlin/issues/12403

SDeath667, 2020-09-10:
I got to this page after having layer shifts for a few days for the first time I ever owned my Prusa i3 MK2S. 
Before I was printing using my Windows pc and Octoprint. I switched to the RPi 3B with Octoprint a month ago. I started seeing the layer shifts on larger prints that lasted over 2-3 hours. 
I checked everything, belts are properly tensioned. 
I found this page, put the GCode on SD and it printed fine. I tried this print for days and it failed at least 7 times using Octoprint. It worked first time from the SD. There seems to be something wrong indeed. I do have some plugins but not the DLP one. 
My printer is connected through a long USB cable that first connected to the PC and later I switched it to the RPi. it is quite long about 3 meters so I suspect this to be the issue in combination with RPi, lower signal power maybe?
I will move the RPi to the printer and use a shorter cable after the Bear conversion I am doing soon. Hopefully this fixes the issue.

Celcius1, 2020-10-13:
I have been getting layer shifts as well and have disabled the following plugins that I have installed
Dashboard 1.15.1
DisplayLayerProgress 1.24.0
M73 Progress 0.2.1
PrettyGCode 1.2.2

for now I'll leave them disabled as my layer shifting has stopped, but I'll re-enable them and do some test prints to see which one may be causing layer shifting.  As one of the shifts I was getting was a subtle shift with each layer.

SDeath667, 2020-10-13:
I bought Repetier Server and that works great. 
Will only use OctoPrint now for mesh bed level visualizer.

f1vefour, 2020-10-13:
> My printer is connected through a long USB cable that first connected to the PC and later I switched it to the RPi. it is quite long about 3 meters so I suspect this to be the issue in combination [...]

Did the cable have any affect?

SDeath667, 2020-10-13:
> Did the cable have any affect?

I didn’t even try to be honest. 
Bought Repetier (it was on discount) and installed that and place the RPi in the enclosure, attached a 5” LCD that worked out of the box and didn’t look back, not one layer shift anymore, but that might have been the same with OctoPrint in a short cable. 
The only thing is the Mesh Bed Level Visualizer I am missing in OctoPrint. 
So I did install it again on my Windows server and will use that to do mesh bed levelling and the nylock mod after I upgrade the printer to a MK3S and full Bear Exxa.

Celcius1, 2020-10-14:
I think it's a firmware issue, as I just put the latest bug fix branch onto my printer and the layer shifting has stopped, I'm going to do a test with the plugins turned back on to see if the issue is as a result of a bug in the firmware.

f1vefour, 2020-10-14:
> I think it's a firmware issue, as I just put the latest bug fix branch onto my printer and the layer shifting has stopped, I'm going to do a test with the plugins turned back on to see if the issue is as a result of a bug in the firmware.

I hope it's not a firmware bug as the printer I'm running hasn't released the sources yet so no way for me to fix this.

Celcius1, 2020-10-14:
Spoke too soon my current print has layer shifted, but it only did it on the current print, looks to be an interment problem. as previous print was fine and I have a bunch of plugins currently disabled

Moorviper, 2020-10-15:
when i read prusa firmware "3.8.1" its no wonder. I also had many problems with that fw.

mrseeker, 2020-10-16:
Question: Did you have a big print and turned on the "g-code visualizer"? I had no issues with my print, I turned that thing on (a 22.5mb big file) and suddenly I notice a noticeable shift on a single layer on my print. Might be a hiccup on the print, might be something else, I don't know...

richardjharding, 2020-12-02:
Just to add some additional anecdotal evidence - I started to recently experience layer shifting in what had been a reliable printer/ Octoprint combination (Mk2 upgraded to Mk2.5)
I checked all the usual causes , belt tension, pulley, ended up swapping steppr and y-axis bearings but I still ended up with layer shifting almost always in y-axis - only cutting speed to 60% seemed to reduce the number and degree of shift
(note shift was not always in same place on print)
Eventually stumbled on this and the DisplayLayerProgress plugin [issue](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/124) 
I first tested a print from SD card at 100% speed (same gcode as before) and it was perfect so I think the hardware, firmware combination is good
Next I tested with DisplayLayerProgress plugin disabled and again print was perfect
Not clear to me why it would have that effect but I'm happy to provide logs, run a test if its of any help

MikePfaff, 2020-12-08:
I have also encountered the same (or similar) problem on my MK3 (FW 3.9.2) with OctoPrint 1.5.1 (OctoPi 0.16.0) while trying to print the new MK3S+ parts. Everything ran fine on an older OctoPrint 1.3.x version (with only Navbar temperature plugin a Prusa ETA override plugin) but started to appear after upgrading to 1.5.x and adding the DLP, Dashboard and Bed Visualizer plugins. I run OctoPrint on a RPi Zero W (being generally aware of its limitations regarding webcams and the effect of WiFi on the CPU load).

I first suspected some kind of mechanical issue because about the same time the problem first appeared i did do some printer upgrades (Misumi bearings/rods, paving stone under the printer, tightening all screws and belts, changing heatbed cable, etc.). After many failed tests, retries and unusable prints, i'd like to report the following observations which might help during your investigation:

- Printing the same file from SD card always works fine.
- 80% of the cases the layer shift happened somewhere around the Z height 1cm mark
- 100% of the cases i got only one layer shift in a print job.
- About 50% of the cases i had the shift in only one of the X or Y axis, the other 50% i had the shift both in the X and Y axis (but even in that case the shift in both X and Y directions happens at the same Z height) 
- Shift is always about 0.5-1mm
- Subjectively seems to happen more often when i have more and/or higher (Z axis) objects and/or larger GCode in a print job. Small objects sometimes print fine.
- I did one try with stealth mode disabled (set through the menu on the printer) and IIRC (not 100% sure though) that did run fine, but unfortunately normal mode is too noisy at night for my my environment.

One thing i was thinking about: It seems that many/most users reporting here did encounter this problem on Prusa machines. I wonder what kind of Raspberry hardware everyone is using and if that maybe plays a role in this issue. Considering that Prusa did recommend the (AFAIK unsupported/not-recommend-with-OctoPrint) RPi Zero W and thus maybe more people are running on that (thinking like "more Prusa = more RPi Zero W").  

I have now temporarily resorted to SD card printing. I might try the following variations and report back when i a have less urgent print jobs to run over the next couple of days:
- OctoPrint in safe mode
- OctoPrint in regular mode without plugins (or without DLP, Dashboard and Bed Visualizer)
- RPi 3 instead of RPi Zero W
- Try normal mode again (as environment permits)

Maybe it's also worth to figure out if there is some "consistency" in the layer shifts among the affected users like e.g. happening at the same mm Z height, same layer count,, same time-since-print-start or similar. Maybe everyone could print the same object and compare if with the same object the problem appears in the same place/height? Maybe it's also worth to compare test prints with the same volume (e.g. a cuboid 1x1x9cm vs 3x3x1cm) among each other and/or in different orientations, to figure out if it has anything to do with volume or with object dimensions on a certain axis.

jneilliii, 2020-12-08:
Definitely try safe mode for testing and disconnect any other external apps from connecting via the api (octodash, octopod, printoid, etc.). There have been previous reports of DLP causing layer shifts which I thought were fixed, but there may still be something there. My plugin (BLV) has been optimized to not process gcode unless flagged to.

Its-my-right, 2020-12-13:
Hi, same here, I recently installed Octoprint via Octopi and I noticed today my first layer shift (Y only). No layer shift on the same model when printing from SD. 
The shift only happens in Y, but on the same model, same Gcode, I got it on two different height each time. But the two times, it was a 0.5-1mm shift. I can provide the model used, it's a drum tom so quite circular, but with some thin vertical ties on the bottom and the top. The two times I noticed the shift, it was on areas where those thin vertical ties are.

My printer is a Prusa MK3S with latest firmware.
Octoprint is up to date, and I'm using several plugin : M73 progress, PrintTimeGenius, DLP, and Octodash, on a raspberry Pi 3 B+.
Sadly it would be difficult for me to run various test (Safe mode, disabling plugin etc) because I'm already troubleshooting something else (differences between models sliced with PrusaSlicer and Cura), so for now I just print from SD.

cp2004, 2020-12-13:
Can I point out the top of this thread:
> **Questions & tasks for everyone running into this issue:**
>
> * rule out mechanical & environmental reasons for your shifts (slipping pullies, loose belts, overheating steppers, undervoltage, ...)
> * [...]

In short, the things that need to happen:
1. Test this in safe mode
2. Remove Display Layer Progress and test.

It's no good to me if this makes you sad, because it makes me *very* sad that people seem to ignore the obvious tests, suggested solutions and more.

Before anyone else thinks about posting in this thread, they should make sure they have the answers to these two questions in the posts:

* **Did the same happen when running OctoPrint in safe mode?**
* **Did the same happen when Display Layer Progress is disabled?**

If the answer to **both** the above is 'yes', then we want to hear about it. If Display Layer Progress is the issue, then we don't want to know about it and you should report it to the *plugin author*. And if you don't want to answer, don't post.

Oh, and if you could answer the rest of the points above then it would be *really* good, but let's not get ahead of ourselves here.

richardjharding, 2020-12-13:
Just to confirm I've been printing in safe mode now
 - perhaps a dozen prints - all perfect with no layer shifting

At first, I thought it was the display layer progress plugin but even disabling that I was still getting the odd shift - running in safe mode has been reliable (also SD card has no issues)(

MikePfaff, 2020-12-13:
> At first, I thought it was the display layer progress plugin but even disabling that I was still getting the odd shift - running in safe mode has been reliable (also SD card has no issues)

For the cases where you had the odd shift even with the Display Layer Progress (DLP) plugin disabled: Was the GCode that you used for these prints freshly created (e.g. newly sent from PrusaSlicer / freshly uploaded to OctoPrint) or was it a GCode that was sent/uploaded to OctoPrint when the DLP plugin was still enabled?

As far as i understand it, the DLP plugin actually modifies the GCode files by injecting M117 commands[1]. So if you start a print job with a GCode file that the DLP plugin has already modified, that could explain why you see layer shifts even with the DLP currently disabled.

An interesting test case would be to use a GCode file that the DLP plugin has touched (DLP plugin was enabled when the GCode was sent to OctoPrint) and which is known to have the layer shift issue and then download that GCode file from OctoPrint onto an SD card and run the print job directly on the printer from the SD card. If the layer shifts still happen, the issue is likely to be either in the way the DLP plugin modifies the GCode file or even in the printer firmware.

By the way: I found an open issue reported against the DLP plugin [#124 DisplayLayerProgress causes layers shifts](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/124) that sounds quite similar to this one.

[1] The M117 commands are used to set the status line text (last one of the 4 lines) on the LCD display. In the Prusa firmware, the M117 command is the first command that the command processor will process before any other command. The comment in the firmware even says "moved to highest priority place to be able to to print strings which includes "G", "PRUSA" and "^"". So i wonder whether the issue might be somehow caused by this high priority handling of M117 commands.

mrseeker, 2020-12-16:
Okay, to confirm some things here:

Running layer shift issues with an Ender3. Have a belt tensioner on both X and Y-axis. Layer shifts on certain areas, and is pretty consistent. Using Marlin 2.0.7.1 with SKR 2.0 hardware.

DLP plugin: Not installed.
M73 plugin: Installed, Running issues.
Octoprint 1.5.2 in safe mode: Running issues.
Still have to test SD card, but since the job is a 25h print I don't want to do it too often. Will do it now for the sake of testing.

Tensioner a bit tighter did help with the shifting, but not that much that I feel confident that it solves the problem. The layer shifting is also consistent and it seems to go back after 3-4 layers. However, when printing a cube the shifting does not seem to occur on those layers, suspecting an issue on the server end rather than on the firmware end.

I might suspect that there is something mechanical at work here, most likely mechanical (acceleration). The weird part is that the layer shift "autocorrects" itself on higher layers, suspecting that there is a rounding error occurring somewhere in the print. Notable transition happens on the 10mm and 35mm mark.

I think the onliest way to confirm that there is nothing with the firmware is to modify the virtual printer to write the received gcode to file, and to compare it with the original.

mrseeker, 2020-12-17:
Update on my last check: I can confirm it is NOT Octoprint that is causing the layer shifts, else I would have seen the same behaviour with my device. This leaves either an issue with the Prusa slicer or the used slicer having rounding issues. The issue is for sure not an Octoprint one, I can't replicate it.

brianblank, 2021-03-31:
I have a Prusa i3 MK3S+ with Raspberry PI 4.  I was also experiencing layer shift with OctoPrint, but not when printing directly from SD Card.  I wanted to share with everyone my solution which so far is holding up.

Early in my build, I configured a Raspberry PI4 with both heat sink and fans in an Ikea Lack Enclosure.  I didn't have any layer shift in my printing then.  I then moved my Raspberry PI4 outside of the enclosure placing it in a new case but without the fans (only with copper heat sinks).  That is when I started to noticed the layer shift.  I changed the case design to add back the fan and the layer shift seems to have went away (I'm still testing though).  However, several days in and still haven't seen any layer shift.

I think what was happening was that the PI4 was overheating.  When a PI4 overheats, it automatically starts to go into thermal throttling lowering the speed of the CPU.  I suspect when it does this, the PI4 can't handle everything and we may be loosing some GCODE causing the layer shift.  I noticed above that people say they are not seeing the layer shift in safe mode.  Most likely they are running fewer plugins which means the CPU probably isn't getting hot enough to go into thermal throttling.

I'm still testing, but several days in and no layer shift yet.

LazeMSS, 2021-03-31:
@brianblank if this is happing it would show up in the logfiles and in the UI if i'm not mistaken. So look for warnings about this.

brianblank, 2021-04-24:
@LazeMSS i'm still not seeing any layer shifts since I added my fan.  I understand what you are saying, but the log is on thr raspberypi,  not  the printer... so it is possible its logging the intent to send a geode but the geode is not making it to the printer.

My recommendation is when someone has this issue, just ask them to include in the stats which model pi they have? If it has a fan or heatsink, and if the pi is inside the enclosure with thr printer (making it hotter).  My experience is that the heatsink alone outside the enclosure was causing layershifts while I did not see the layer shift problem with the fan outside the enclosure.

LazeMSS, 2021-04-25:
@brianblank  i'm 99% certain if the issues are having is dues to heat / missing cooling then the vcgencmd would should throttling.

mrseeker, 2021-04-25:
The soft limit (at which the raspberry pi starts throttling) is 60c. Best to check if your raspberry reaches temperatures over 55c during printing, this might be the reason for throttling...

brianblank, 2021-07-24:
@LazeMSS   After several months, I still have not seen another layer shift since adding a fan to my Raspberry PI 4.  Before adding the fan, I was getting layer shifts every other print.  I'm nearly certain the fan resolved my issue.  I'd be curious to know if people are seeing the layer shift after they move the PI outside the enclosure and also add a fan to it.

mrseeker, 2021-07-24:
Hi Brian,

I changed my enclosure to be less warm, and that does help. It might be
related to thermal throttling, however it's very hard to say if that is the
case because it's also very hard to verify as the source of the issues.

[6 more replies]