OctoPrint/OctoPrint #3527: [Request] Mobile Interface Improvement

tomfarnham, 2020-04-19:
The utility of the interface on mobile devices is a little cumbersome. The project already uses bootstrap for a CSS framework, which when implemented correctly will render mobile interfaces. My main concern with the interface is pushing the wrong button can have negative side-effects, like crashing the print head.

The main issues that I've noticed:

- Controls are too small and close together, requiring the user to zoom in.

- Zooming in gives you a 50/50 chance of clicking the button you want or panning the screen. 

Ideally, the correct measures in the style would be implemented like using grids with appropriate size scaling. This would eliminate the white-space margins on the sides of the screen, and stack the panes within the interface instead of putting them side-by-side. Additionally, buttons can be scaled for user-interface base on the screen size.

Additionally, menu style options can be used instead of toolbars and the like leading to an interface that is easier and more reliable to use.

foosel, 2020-04-19:
A rewrite of the UI is on the roadmap for 1.5.0 which will implement a proper modern responsive design. Adapting the current UI is not feasible: the currently used version of bootstrap is long outdated and no longer supported, but cannot be updated as that would break compatibility for the ecosystem of existing plugins. THe way forward is a rewrite that for a while will live concurrently alongside the existing UI.

I created the ability to completely replace the UI of OctoPrint two years ago by plugins in the hopes someone would step up to create a mobile UI plugin, but sadly that hasn't happened. In the meantime I suggest to use one of the available third party clients or the TouchUI plugin (a reskin, not a replacement).

tomfarnham, 2020-04-20:
Awesome, keep up the good work. I'll see if I can contribute in the future.

cp2004, 2021-03-31:
Since this is still on the roadmap and intended to be implemented, I'd leave this open, it can serve as a reminder for anyone looking to open more issues 🙂

Jakobud, 2021-10-20:
Ya I think the best approach would be to also include Bootstrap 2, but that could have some weird effects including Bootstrap 2 and Bootstrap 5 CSS on the same page. 

The team would just need to come to a decision on how long they would keep the Bootstrap 2 CSS in there. Give plenty of time for authors to upgrade their plugins and eventually just pull the plug on Bootstrap 2.

Jakobud, 2021-10-20:
Actually thinking a little more about this, it wouldn't work very well to include both Bootstrap 2 and Bootstrap 5 at the same time. The problem is this:

You want BS5 styles to be prioritized over BS2 styles so that the main application interface looks good.

But the whole point of including BS2 is that so old plugins can still use those styles. But the problem is that they wont be using them because BS5 styles are going to be taking priority. Most of the primary layout CSS classes are the same between BS2 and BS5 (`container`, `row`, etc).

Honestly maybe the best approach is to simply replace BS2 with BS5 and just leave it. Plugins just gotta update. There really isn't an easy way to include both libraries and force the main UI to use BS5 and allow plugins to still use BS2 by default.

cp2004, 2021-10-20:
It's not something that we're going to do soon - and there's plans to move away from bootstrap entirely, and actually re-write the way it works - if we are going to introduce such huge breaking changes, then we might as well combine converting the UI to be written in React (or similar) with updating the design. 

Rather than replace Bootstrap, force every plugin author to rewrite their plugins' UI (which would also break compatibility for older versions, not everyone updates at the same time - huge mess) and then rewrite the whole thing again, forcing all the plugin authors into complete rewrites it makes more sense to do it all at once.

There's a *lot* more to it than 'just include bootstrap 5'. I think nearly every single class name has changed, the structure of the HTML needed has changed - it's impossible to try and make a compatibility layer, I've tried a few different experiments.

LazeMSS, 2021-10-20:
@cp2004  and I have both spent a fair time on looking into it and yes the overhead and time spent on making anything work without work for the plugin authors is basically not possible. So for now I would recommend UI Customizer and as Charlie wrote the frontend needs to be rewritten.

Jakobud, 2021-10-20:
Gotcha.

I'm working on getting a dev environment setup. Can someone tell me how these .less files are compiled to .css? I'm not seeing any sort of build setup for this.

gabrielcsapo, 2022-07-04:
Is there an RFC or document defining what the UI expectation is for 2.0? I am assuming we would need to have a way for existing plugins to work with the redesign, but maybe there is a 1.0 and 2.0 plugin compatibility cut-over?

jneilliii, 2022-07-04:
There definitely will be a compatibility cut over process because the underlying frameworks are going to be changing. I don't think there has been any discussion on any type of layout changes, but the framework is going to be react/modern ui if I'm not mistaken.

gabrielcsapo, 2022-07-05:
Does it make sense to make an issue specifically for the 2.0 interface design / check list?

cp2004, 2022-07-05:
> Does it make sense to make an issue specifically for the 2.0 interface design / check list?

At the moment, possibly not because the project is not really in a situation where we could look at it & move forward with it currently. There's more things on the list that need to be tackled first.

The major sticking point is a plugin system. We considered React & it is still the current intention but so far there is no clear way to create a plugin system with the way the builds work.

ivebeenlinuxed, 2023-06-02:
I wonder if it would be possible to use Angular and Module Fedoration...

I've started a branch to try put angular into OctoPrint. It's a mess, and it is not my commitment to see it through, but seemed an interesting idea to experiment with. I'm sorry, but in general I can't commit much time to this - and I'd rather be upfront about it.

My musings have currently led me to this:
https://github.com/ivebeenlinuxed/OctoPrint

To run use:
```
cd src/octoangular/;
ng serve --watch --hmr;
``` 
Run octoprint on the same machine on port 5000. The file src/octoangular/src/proxy.conf.json will proxy the octoprint API through the Angular webserver.

So far:
- The login screen detects "first run" and shows a registration screen
- The registration screen uses the API to register the first user and then marks the setup wizard as complete (no other steps)
- The login screen allows you to use this user to login (Some fix-ups with CSRF tokens)

Questions:
- What does the development environment environment look like for a typescript/python/flask application?
- What would a build pipeline look like? (At the moment for octoprint you just "git clone" and the application is [nearly] ready to run, whereas Typescript needs building before running). The project would need to build and host releases.
- How does the API need to change (OAuth2? Continue with cookies? CRSF?)
- What would a plugin look like? (https://www.thirdrocktechkno.com/blog/how-to-build-micro-frontend-in-angular-with-module-federation/)
- What benefits should such an improvement target? (themes? mobile? multiple API backends in a single interface [multi-printer console]? Sandbox of plugin dependencies (think plugins have their own bootstrap version vs global version?). I'm not a UX/UI designer, so might be best finding someone who can do some concepts if there are such contacts.

jneilliii, 2023-06-02:
Do plugins work? That's the problem with changing the core and it's been discussed in great detail before that this is what's holding back OctoPrint. If you want more mobile friendly try UI customizer plugin, it has a mobile responsive option you can enable.

jneilliii, 2023-06-02:
The other option would be to create a UIPlugin mixin implementation.

ivebeenlinuxed, 2023-06-02:
Hi @jneilliii 

This commit adds a simple plugin manager. It allows for plugins to have an Angular Module which is dynamically loaded into the main interface. As I say, a Proof of Concept only.

https://github.com/ivebeenlinuxed/OctoPrint/commit/5bde8022e81e5f1c09c8b3c86153e9ef33191613#diff-f0591604d6bbdc2c682cdb8203fb7272e3d3685d65cb4d18d5064c8e0e4a0dcf

foosel, 2023-06-05:
So, module federation is something I (ironically) fell over *after* wrangling webpack into submission to allow for something similar in a React based prototype. So that's definitely something on the menu. 

But Angular is something I'd rather not use here. I've heard some bad things about it from devs, and if we just go by [the numbers](https://www.jetbrains.com/lp/devecosystem-2022/javascript/) (popularity and ecosystem is kinda important to attract contributors and not make plugin devs despair, a lesson I learned from choosing the very much niche KnockoutJS for the core UI) React is a better bet. I don't have any Angular experience at all, so I have to judge by what I heard about it, but that's another factor given you already said you'd not be able to really contribute here. 

So, thank you very much for showing us this, feel free to experiment further and by all means look into the UiPlugin mixin if you want (OctoPrint's core UI doesn't have to be the only UI, and I've been desperately waiting for someone to make use of the option to build on it!), but for the future of OctoPrint I think Angular would not be the right choice.

ivebeenlinuxed, 2023-06-05:
Hi @foosel

Thanks for taking the time to reply in depth. Certainly understand your reasoning above. In fact, I was using the OctoPrint project purely to trial out Module Fedoration within an established system.

From the documentation, Angular seems a much larger framework from React (though I have no experience in React in practice!), and although Angular is supported by some big names I can certainly understand how it could affect the plugin architecture to have a less well supported framework. React may well be a good choice alongside Python too, where it might well allow for lighter plugins and faster development at the expense of some mandated structure and bundled features.

I'm sure that as both frameworks share a webpack core, there is probably a fair amount of overlap between the solutions here and the way Module Fedoration and Flask/Pyton can be used to generate manifests of plugins to be dynamically imported. When an issue is created for React I would certainly attempt to find some time to help your cause.

In any case, thank you for your hard work in keeping OctoPrint running! Absolutely love the software.

foosel, 2023-06-06:
Thanks for understanding, I always hate shooting down good work that has been done already, but sometimes it has to be done for the sake of the project's direction 😅

Just in case you are curious, my experiments with a plugin enabled react frontend can be found here: 

- https://github.com/OctoPrint/OctoPrint-UiPrototype
- https://github.com/OctoPrint/OctoPrint-UiPrototype-TestPlugin

However, that does not yet utilize module federation (as I said, I came across that AFTER I did it the hard way), so I guess I'll need to pencil experiments in that direction in for another action week in the near future.

ivebeenlinuxed, 2024-07-06:
Hi @foosel! Any more thoughts on this? Do you still have plans to go React/Angular? Your webinars seem to suggest things are pretty busy at the moment, so understanding V2 might be a little too much to think about right now!

Thanks again for all your hard work!

foosel, 2024-07-17:
Absolutely NO bandwidth at all for this at the moment and the foreseeable future.

ivebeenlinuxed, 2025-03-10:
Hey @foosel. I have some reasons to find out a little more about React, so thinking of doing a "octoreact" plugin, where I can play around with React a bit and get a hang of it. From our previous messages it seems that module fedoration may be something to continue messing with.

Obviously there is a massive questions over what the aims of a new UI are, but as I'm sure that's a pretty heavy question, I'm just going to stick to testing React and Module Fedoration.

Let me know if there is anything you think there is anything you'd be interested in me using/exploring. As an example, I was going to go Bootstrap (because it's the way I've always done it ;-) ) I did see some people have a some conversations about material design instead...