OctoPrint/OctoPrint #3530: SIGTERM not firing Shutdown event

OutsourcedGuru, 2020-04-21:
#### What were you doing?
I'm currently working on a plugin to allow an Xbox controller to control an OctoPrint instance. The plugin needs to thread some code using `sudo` which then makes it harder to cleanly kill that afterwards. The driver needs to run as `root` whereas `pi` is the standard user for OctoPrint.

After a standard OctoPrint shutdown I was finding that the spawned `xboxdrv` executable was still running. Subsequent service starts then would complain that the new `xboxdrv` instance would fail to talk to the USB receiver device driver.

I both tried the `ShutdownPlugin` mixin as well as the `EventHandlerPlugin`. Neither received the event from a SIGTERM-derived shutdown. Both received the event from a web interface—induced shutdown/restart action, however.

#### What did you expect to happen?
When I stop the OctoPrint service using the external service handler, I expected the `Shutdown` event to bubble up into my plugin via either provided mixin option.

#### What happened instead?
When exercising the web interface -> system menu -> Restart OctoPrint option the Shutdown event is fired off. When running `sudo service octoprint stop` or `pkill octoprint` the event is not fired off.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Work-around
I was able to patch OctoPrint's octoprint/server/\_\_init\_\_.py [here](https://github.com/OctoPrint/OctoPrint/blob/97a82c0cc13bcf936cdd723162ea7cccfb0dccf4/src/octoprint/server/__init__.py#L923) so that this would work for me as expected:

```
		def sigterm_handler(*args, **kwargs):
			on_shutdown() # Added only this line to OctoPrint's code
			# will stop tornado on SIGTERM, making the program exit cleanly
```

Then in my code, I added the `octoprint.plugin.EventHandlerPlugin` mixin in the main class. I then trapped the `Shutdown` event to kill my rogue Xbox thread so that it wouldn't ghost after an external OctoPrint shutdown. I'm sure I could have instead added the ShutdownPlugin mixin and triggered from the `on_shutdown()` function as well.

#### Version of OctoPrint
OctoPrint 1.4.0, OctoPi 0.17.0

#### Operating System running OctoPrint
OctoPi 0.17.0, Raspbian Buster Lite

#### Printer model & used firmware incl. version
Robo C2, Robo 3D's fork of Marlin

#### Browser and version of browser, operating system running browser
Safari 12.1.1 and macOS Mojave 10.14.5

#### Link to octoprint.log
```
# This is my promiscuously-logging on_event() handler to log all events seen
2020-04-20 09:39:48,711 - octoprint.plugins.xbox - INFO - on_event FirmwareData
...
# I then sudo service octoprint stop from the command line:
2020-04-20 09:40:21,082 - octoprint.server - DEBUG - SIGTERM received...
2020-04-20 09:40:21,083 - octoprint.server - DEBUG - Shutting down tornado's IOLoop...
2020-04-20 09:40:21,084 - octoprint.server - DEBUG - Tornado's IOLoop stopped
```

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

OutsourcedGuru, 2020-04-21:
In a way, this is essentially the same [request](https://github.com/OctoPrint/OctoPrint/issues/3027) from a while ago.

kantlivelong, 2020-04-21:
Well I think I found why I couldn't replicate it initially. Seems like this may be an issue with OctoPrint running on Python 2.

**Python 3**
```
2020-04-21 19:24:43,656 - octoprint.server - DEBUG - SIGTERM received...
2020-04-21 19:24:43,656 - octoprint.server - DEBUG - Shutting down tornado's IOLoop...
2020-04-21 19:24:43,657 - octoprint.server - DEBUG - Tornado's IOLoop stopped
2020-04-21 19:24:43,657 - octoprint.server - INFO - Shutting down...
2020-04-21 19:24:43,881 - octoprint.plugin - DEBUG - Calling on_shutdown on discovery
2020-04-21 19:24:43,882 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-04-21 19:24:43,882 - octoprint.events.fire - DEBUG - Firing event: Shutdown (Payload: None)
2020-04-21 19:24:43,882 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7fb89c1af710>>
2020-04-21 19:24:43,883 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-04-21 19:24:43,883 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-04-21 19:24:43,883 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-04-21 19:24:43,883 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-04-21 19:24:43,883 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-04-21 19:24:43,884 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-04-21 19:24:43,884 - octoprint.events - INFO - Event loop shut down
2020-04-21 19:24:43,890 - octoprint.plugin - DEBUG - Calling on_shutdown on tracking
2020-04-21 19:24:43,890 - octoprint.server - INFO - Goodbye!
```

**Python 2**
```
2020-04-21 19:29:06,595 - octoprint.server - DEBUG - SIGTERM received...
2020-04-21 19:29:06,596 - octoprint.server - DEBUG - Shutting down tornado's IOLoop...
2020-04-21 19:29:06,596 - octoprint.server - DEBUG - Tornado's IOLoop stopped
2020-04-21 19:29:06,597 - octoprint.server - INFO - Shutting down...
2020-04-21 19:29:07,404 - octoprint.plugin - DEBUG - Calling on_shutdown on discovery
2020-04-21 19:29:07,404 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-04-21 19:29:07,405 - octoprint.events.fire - DEBUG - Firing event: Shutdown (Payload: None)
2020-04-21 19:29:07,408 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-04-21 19:29:07,409 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-04-21 19:29:07,410 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-04-21 19:29:07,412 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-04-21 19:29:07,413 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-04-21 19:29:07,413 - octoprint.plugin - DEBUG - Calling on_shutdown on tracking
2020-04-21 19:29:07,414 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-04-21 19:29:07,415 - octoprint.events - INFO - Event loop shut down
2020-04-21 19:29:07,417 - octoprint.server - INFO - Goodbye!

```

kantlivelong, 2020-04-21:
Never-mind I'm clearly blind. Both environments are calling on_shutdown for me but it wasn't in the same order so I missed it :facepalm: .

kantlivelong, 2020-04-21:
@OutsourcedGuru Could you ensure that `DEBUG` is logging is set for the classes listed above in my logs then repost a full clean log?

DanielJoyce, 2020-05-07:
If you set your XBOX controller thread to be a "Daemon" thread, it won't prevent octoprint from shutting down. That is the proper fix to avoid.

If you are spawning a seperate process (xboxdrv), then you should configure a process group. All members of said group can then be set to cleanly exit.

https://pymotw.com/2/subprocess/

As for the 'driver' needing to run as root, that can be fixed as well with some udev magic

```
# put in /etc/udev/rules.d/90-xbox-controller.rules
# Some hw device a user needs access too
SUBSYSTEMS=="usb", ATTRS{idVendor}=="XXXX", ATTRS{idProduct}=="yyy", TAG+="uaccess"
```
use `lsusb` to find the idVendor:idProduct values for the connected device.

`uaccess` is a magical facl applied to the generated /dev/ entry that allows the currently logged in SEATED user ( the one using the current active X/Wayland session ) to use the device. No need for xboxdrv to run as root anymore. A lot of controllers and devices like keyboards and mice use this rule, you just need to add a matching entry for your controller. When you log out, the permission goes away. If someone else logs in, or if two people are logged in, when you switch X displays, the person active gets access. It all works pretty well.

```
# Restart udev
# sudo udevadm control --reload-rules && sudo udevadm trigger
```

kantlivelong, 2020-05-07:
> If you set your XBOX controller thread to be a "Daemon" thread, it won't prevent octoprint from shutting down. That is the proper fix to avoid.

Are you seeing this in code somewhere?

kantlivelong, 2020-05-12:
@OutsourcedGuru just following up. Wanted to tackle this if it is indeed an issue.

jneilliii, 2021-07-23:
I think I'm running into the same issue with one of my plugins. Related issue [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/274). After some further testing on my end it seems that not all the on_shutdown callbacks are completing prior to the pi powering off. 

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6866390/octoprint.log)