OctoPrint/OctoPrint #3532: Something crashed inside the serial connection loop: error: can't start new thread

rudnerbjoern, 2020-04-22:
### What were you doing?

Printing a job from the Cura to Octoprint on Octopi.

Somewhere in the middle of the job octoprint stopped with an error message.
The printer was "idle" since then hanging at the same position, still heating the bed and the extruder.

1. Send job from Cura to octoprint
2. wait (job run for some hours)

#### What did you expect to happen?

Expected the job to be finished.

#### What happened instead?

The job stopped in the middle with this error:

2020-04-22 03:38:10,680 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 69663, current line = 69667
| Last lines in terminal:
| Recv: ok N69652 P1 B3
| Send: N69658 G1 X119.736 Y180.264 Z108.15 E1881.95726*50
| Send: N69659 G1 X114.193 Y177.468 Z108.154 E1882.26699*15
| Recv: ok N69653 P1 B3
| Recv: ok N69654 P0 B3
| Send: N69660 G1 X112.938 Y171.398 Z108.158 E1882.57623*15
| Send: N69661 G1 X108.308 Y167.274 Z108.163 E1882.88557*2
| Recv: ok N69655 P0 B3
| Send: N69662 G1 X108.665 Y161.076 Z108.167 E1883.1953*53
| Recv: ok N69656 P0 B3
| Send: N69663 G1 X105.262 Y155.884 Z108.171 E1883.50501*0
| Recv: ok N69657 P0 B0
| Send: N69664 G1 X107.21 Y150 Z108.175 E1883.81423*38
| Recv: ok N69658 P0 B0
| Send: N69665 G1 X105.264 Y144.116 Z108.179 E1884.12342*10
| Recv: Not SD printing
| Recv: ok N69659 P0 B0
| Send: N69666 G1 X108.67 Y138.926 Z108.183 E1884.43313*48
| Recv: Error:checksum mismatch, Last Line: 69662
| Recv: Resend: 69663
2020-04-22 03:38:10,691 - octoprint.plugins.influxdb - INFO - Using existing database `openhab_db`
2020-04-22 03:38:10,704 - octoprint.util.comm - INFO - Ignoring resend request for line 69663, that still originates from lines we sent before we got the first resend request
2020-04-22 03:38:10,706 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 1781, in _monitor
    self._handle_resend_request(line)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3020, in _handle_resend_request
    self._resend_ok_timer.start()
  File "/usr/lib/python2.7/threading.py", line 736, in start
    _start_new_thread(self.__bootstrap, ())
error: can't start new thread
2020-04-22 03:38:10,748 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: See octoprint.log for details)"

#### Did the same happen when running OctoPrint in safe mode?

I was not able to rebuild the sutiation causing the problem.

#### Version of OctoPrint

OctoPrint 1.4.0 auf OctoPi 0.17.0

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Creality CR 10 S Pro V2
TM3D Marlin "2.0.5_DW6.1"

#### Browser and version of browser, operating system running browser

Google Chrome Version 81.0.4044.113

#### Link to octoprint.log

https://gist.github.com/rudnerbjoern/665602d7092fc4168c2e941d0ac3fbec

#### Link to contents of terminal tab or serial.log

| Last lines in terminal:
| Recv: ok N69652 P1 B3
| Send: N69658 G1 X119.736 Y180.264 Z108.15 E1881.95726*50
| Send: N69659 G1 X114.193 Y177.468 Z108.154 E1882.26699*15
| Recv: ok N69653 P1 B3
| Recv: ok N69654 P0 B3
| Send: N69660 G1 X112.938 Y171.398 Z108.158 E1882.57623*15
| Send: N69661 G1 X108.308 Y167.274 Z108.163 E1882.88557*2
| Recv: ok N69655 P0 B3
| Send: N69662 G1 X108.665 Y161.076 Z108.167 E1883.1953*53
| Recv: ok N69656 P0 B3
| Send: N69663 G1 X105.262 Y155.884 Z108.171 E1883.50501*0
| Recv: ok N69657 P0 B0
| Send: N69664 G1 X107.21 Y150 Z108.175 E1883.81423*38
| Recv: ok N69658 P0 B0
| Send: N69665 G1 X105.264 Y144.116 Z108.179 E1884.12342*10
| Recv: Not SD printing
| Recv: ok N69659 P0 B0
| Send: N69666 G1 X108.67 Y138.926 Z108.183 E1884.43313*48
| Recv: Error:checksum mismatch, Last Line: 69662
| Recv: Resend: 69663

#### Link to contents of Javascript console in the browser

UI was not involved in the error

#### Screenshot(s)/video(s) showing the problem:
no screenshot available

I have read the FAQ.

DanielJoyce, 2020-05-07:
```The “can't start new thread” error almost certainly due to the fact that you have already have too many threads running within your python process, and due to a resource limit of some kind the request to create a new thread is refused.```

Seems like the process is starting tons of new threads, and not reaping or forgetting to periodically call join to reap them.

kantlivelong, 2020-06-02:
- What is with all the disconnects/reconnects by **octoprint.plugins.influxdb**?
- Is this issue still happening? If so, does it occur without that plugin?

Going to go out on a limb here and say that the issue is likely caused by that plugin or one of its dependencies.

foosel, 2020-06-12:
ping @rudnerbjoern, we need answers here. Agreeing with @kantlivelong that it's probably caused by a plugin.

rudnerbjoern, 2020-06-12:
I can confirm that there was a problem with the influxdb. There was some limit in place about the total amount of events to be written to the database within a timeperiod.
However the plugin did not respect the server message and tried to write more events even so it was not possible to fulfill these request.

Up to now I have disabled the influxdb plugin and have not had any errors like this again.

foosel, 2020-06-18:
Closing this then, thanks for reporting back :)

closed 2020-06-18 by foosel

OevreFlataeker, 2020-09-02:
@rudnerbjoern How did you eventually find out that and which plugin was the root cause of the issue in the end? I am currently experiencing a similar issue after letting octoprint run for a long time without a power-on printer (printer is only used from time to time but the raspberry pi is running 24/7). 

The stacktrace is different but the root cause ("Cannot start new thread") is the same here:
https://community.octoprint.org/t/exception-on-api-connection-post/23732/2

Thank you

OevreFlataeker, 2020-09-02:
I can definitely state that the errors will for sure be based on the fact that the pi runs out of memory... this is what happens when I power off the printer.... and print the number of threads of the main process after a certain amount of time
```
pi@octopi:~ $ ps aux | grep 19603

pi       19603 21.5  7.6 691316 68136 ?        Sl   15:16  39:16 /home/pi/oprint/bin/python2 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000

pi@octopi:~ $ date && ps hH -p 19603 | wc -l
Wed  2 Sep 18:17:44 CEST 2020
53
pi@octopi:~ $ date && ps hH -p 19603 | wc -l
Wed  2 Sep 18:17:47 CEST 2020
53
pi@octopi:~ $ date && ps hH -p 19603 | wc -l
Wed  2 Sep 18:17:51 CEST 2020
53
pi@octopi:~ $ date && ps hH -p 19603 | wc -l
Wed  2 Sep 18:17:55 CEST 2020
54
pi@octopi:~ $ date && ps hH -p 19603 | wc -l
Wed  2 Sep 18:18:49 CEST 2020
73
pi@octopi:~ $ date && ps hH -p 19603 | wc -l
Wed  2 Sep 18:20:48 CEST 2020
113
```

You can clearly see where this is leading after some hours... question is now how to find the route cause

jneilliii, 2020-09-02:
My assumption this is coming from a third-party plugin or external application that is requesting API information. I tend to see a lot of noise in my octoprint.log from OctoDash when my printer is off because it's constantly polling the API to see if the printer is alive or not.

OevreFlataeker, 2020-09-02:
I am recently using octoscreen to have the UI on a seperate 7" TFT as well. How could I diagnose this? Can I influence the log level of the octoprint.log to catch those? It seems it is then not actually related to the powered off printer but will happen either way?
I noticed that the problem starts appearing right away after I restart octoprint: I can see how the number of threads are increasing... Like around 20 after restart and 70 after 3,4 minutes...

OevreFlataeker, 2020-09-02:
OK it seems I found what I was looking for :-(

https://github.com/Z-Bolt/OctoScreen/issues/114

OevreFlataeker, 2020-09-02:
From what I read the Octoscreen devs see the problem on the Octoprint side, because its Octoprint which is accumulating the threads and the problem does not go away when Octoscreen is killed. To sum (https://github.com/Z-Bolt/OctoScreen/issues/56) up: The assumption there is, that Octoprint is not tearing down the REST connections fast enough or at all, when Octoscreen does its polling. So the question is: has someone already looked into that on the Octoprint side? Haven't seen an issue for this. Shall I open one?

foosel, 2020-09-02:
Sounds familiar to #3376. And that has been solved as far as is possible from OctoPrint's side. OctoScreen should not blindly spam connection requests. Not only can this cause issues still apparently if it happens fast enough, it's also just wrong to spam connection requests outside of the user's control and could weak havoc with ongoing SD card based print jobs for example.

OctoScreen should just query the printer status and not try to connect again and again and again here, especially not of the connection attempts are obviously constantly failing.

OevreFlataeker, 2020-09-02:
Thank you for commenting. I agree #3376 seems to be the same issue. Sorry for not finding that reference before, I was searching for the particular "cannot start new thread" error scheme.
I understand all that has been done from your side has been summarized in https://github.com/OctoPrint/OctoPrint/issues/3376#issuecomment-645984328 and as thus should already be available in my running instance (v.1.4.2)? Or would I need to check out another branch or something in order to benefit from those fixes?

So having heard the arguments on both sides I understand that octoprint is somewhat prone to a "DoS" attack (to call it like that), however the overall attack surface is very little and in this case the behavior of OctoScreen is considered not best practice by polling over and over again.

I assume a proposed fix could be: "Try to get the connection data and if it fails, stop it after like 10 tries and only retry when user pressed a button to do so". However what I don't understand yet is, if OctoScreen under normal conditions, i.e. printer is connected and all, causes a "DoS" too, by repeatedly polling the status of all the relevant parameters or whether this is something which will gracefully be handled and won't cause any problem?

foosel, 2020-09-03:
> I understand all that has been done from your side has been summarized in [#3376 (comment)](https://github.com/OctoPrint/OctoPrint/issues/3376#issuecomment-645984328) and as thus should already be available in my running instance (v.1.4.2)?

Yes.

> I assume a proposed fix could be: "Try to get the connection data and if it fails, stop it after like 10 tries and only retry when user pressed a button to do so".

The problem is, that request is not trying to get the connection data, it's trying to create a new connection. It shouldn't be doing that 10 times in a row, it shouldn't be doing that on its own at all frankly, it should only be doing that on user prompt. 

In any case, I suggest to move the conversation over to the linked above forum thread, as this ticket here is unrelated to OctoScreen (the culprit here was a misbehaving influxdb connection) and the one related to it has been handled as far as possible from OctoPrint's side of things.

OevreFlataeker, 2020-09-03:
Thank you, will do so. 
FYI: I cloned the Octoscreen repo and added as a first step a config option to define the poll interval. Now set to 10s instead of 2s and testing whether the problem re-emerges with that interval as well. I understand this is not the best solution, and also didn't check any side-effects of this. However with my lack of Go experience (actually this experience is 2h old :-D ), it is the best I can do for the moment.