OctoPrint/OctoPrint #3536: [Potential Bug]: plugin settings set to null instead of being removed from config.yaml

jneilliii, 2020-04-26:
#### What were you doing?

Trying to debug issues with end users on plugins and requested them to use the new cleanup plugin data button in Plugin Manager. 

#### What did you expect to happen?

The entire `plugin_identifier` section to be removed from all areas of config.yaml.

#### What happened instead?

After completing the above steps there was still a `plugin_identifier` in the plugins section of config.yaml but it was set to a value of null and remained in a couple of other places, like this.

```
plugins:
    plugin_identifier: null
    softwareupdate:
        check_providers:
            plugin_identifier: plugin_identifier
server:
    seenWizards:
        plugin_identifier: null
```

#### Did the same happen when running OctoPrint in safe mode?

This is a problem requiring OctoPrint to not be in safe mode in order to do this.

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 81.0.4044.122 (Official Build) (64-bit)

#### Link to octoprint.log

I have not been able to reproduce this issue personally, but had to 2 people have the same thing happen [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/195#issuecomment-609033705) and [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/195#issuecomment-619424910).

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

No errors

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

kantlivelong, 2020-05-12:
I was able to reproduce by running cleanup twice. What I did notice is that `octoprint.plugins.pluginmanager._cleanup_settings` is calling `global_set` when I would think it makes more sense to just call `global_remove`. Additionally `octoprint.plugins.pluginmanager._cleanup_settings` doesn't appear to be doing anything with the `seenWizards` section.

I submitted #3566 which should fix this. I did not include clearing out any settings under `appearance` as it *shouldn't* really affect functionality but that is debatable.

foosel, 2020-05-15:
That's fixed by the above PR (thanks for that @kantlivelong!) and ready for 1.4.1

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel