OctoPrint/OctoPrint #3537: [BUG] Force Check for Updates determines there's nothing to Update (but it's false)

bolsoncerrado, 2020-04-27:
#### What were you doing?

Force a Check for Software Updates

#### What did you expect to happen?

Octoprint would report Xxxx plugins to be updated (more specifically RequestSpinner)

#### What happened instead?

Octoprint reported Everything is up to date.

#### Did the same happen when running OctoPrint in safe mode?

Didnt try as Plugin functionality is disabled on safe mode.

#### Version of OctoPrint

OctoPrint 1.4.0 running on OctoPi 0.16.0

#### Operating System running OctoPrint

OctoPrint 1.4.0 running on OctoPi 0.16.0

#### Printer model & used firmware incl. version

Prusa MK3S

#### Browser and version of browser, operating system running browser

MacOS 10.14.6

#### Screenshot(s)/video(s) showing the problem:

[image]

Note the right frame showing an identical instance of octoprint (same IMAGE file) running on another IP on the same LAN and having updated this morning just fine).

Have tried to reboot and no change.

I have read the FAQ.

foosel, 2020-04-27:
@bolsoncerrado why do I need to tell you every time that **I. NEED. LOGS.**? 😕

bolsoncerrado, 2020-04-27:
Wooops! Sorry!
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/4538400/octoprint.log)

foosel, 2020-04-27:
I don't see it checking for updates in there.

bolsoncerrado, 2020-04-27:
What can I say? You see the screenshot?

foosel, 2020-04-27:
Was that from this morning?

foosel, 2020-04-27:
Also: Does it only affect RequestSpinner or other plugins as well? From your screenshot it looks like network issues not allowing it to fetch data about new releases (note the "-" on your second screen where it should have the available version).

bolsoncerrado, 2020-04-27:
> Was that from this morning?

YEs this exact morning, but last week I had a similar behaviour with another plugin on the same Pi.

bolsoncerrado, 2020-04-27:
> Also: Does it only affect RequestSpinner or other plugins as well? From your screenshot it looks like network issues not allowing it to fetch data about new releases (note the "-" on your second screen where it should have the available version).

You mean I can operate on the LAN but not on WAN with this Pi? Lemme do a quick check.

bolsoncerrado, 2020-04-27:
AFAI can see, Pi do have access to the Internet...

foosel, 2020-04-27:
I cannot reproduce this here. Works just fine on my end. 

In general the update check requires a) existing releases on Github/PyPI/whatever is configured to check and b) a working network connection there. If the Software Update for whatever reason cannot determine the current version (network error, no releases published or something like that) it will assume everything is up to date and report that accordingly. So from your issue description so far and considering that I cannot reproduce it locally on my installs here, I'm assuming there's been some network hiccup somewhere between you and github's servers that caused this. I suggest you try again in a couple hours, and if the issue persists report back, otherwise close this issue.

foosel, 2020-08-04:
Closing due to lack of response.

closed 2020-08-04 by foosel

bolsoncerrado, 2020-12-04:
Came here as this issue still persists, at least on my setup. We don't use any fancy router nor switch with web management, just the usual plug, play and forget kind of ethernet devices.

So far all Pis have access to the internet, but for whatever reason I can only update a few of them and the issue seems random. I've tried to reboot on safe mode and retry to force the update to no avail and all its combinations but it's simply ignoring me.

I know you closedthis on 4Aug. but on 8 Oct. you labelled this as BUG.

If there's any sort of instructions I can run on the affected PI(s) to help troubleshoot this, lmk and I'll gladly do my best to help!

This time, obviously, was trying to upgrade to 1.5.1 where as some Pis suceeded and some others just didnt notice the 1.5.0 update and still remain on 1.4.2....

cp2004, 2020-12-04:
Hint: As per above, the `octoprint.log` would be helpful 😉.

bolsoncerrado, 2020-12-04:
She didnt notice any updates check ie made no difference last time but ok...

cp2004, 2020-12-04:
I am using 1.5.1 (well, maintenance branch) but the failures are quite clearly logged:
```

2020-12-04 16:51:07,523 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2020-12-04 16:51:07,737 - octoprint.plugins.softwareupdate - WARNING - Unknown update check type for target swupdate_test: <n/a>
2020-12-04 16:51:07,907 - octoprint.plugins.softwareupdate - WARNING - Could not check eeprom_marlin for updates due to running into a rate limit: Github rate limit exceeded, reset at 2020-12-04 17:43
2020-12-04 16:51:07,915 - octoprint.plugins.softwareupdate - WARNING - Could not check file_check for updates due to running into a rate limit: Github rate limit exceeded, reset at 2020-12-04 17:43
2020-12-04 16:51:08,037 - octoprint.plugins.softwareupdate - WARNING - Could not check themeify for updates due to running into a rate limit: Github rate limit exceeded, reset at 2020-12-04 17:43
2020-12-04 16:51:08,041 - octoprint.plugins.softwareupdate - WARNING - Could not check touchui for updates due to running into a rate limit: Github rate limit exceeded, reset at 2020-12-04 17:43
2020-12-04 16:51:08,058 - octoprint.plugins.softwareupdate - WARNING - Could not check virtual_printerconfig for updates due to running into a rate limit: Github rate limit exceeded, reset at 2020-12-04 17:43
2020-12-04 16:51:09,125 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
```

jangrewe, 2021-01-22:
I have two Pis that are set up absolutely identical. Same plugins installed, same OctoPrint version, same configs, everything. I haven't used them in a while, but today it wanted to, and one of them happily announced all the updates and releases and blog posts and new plugins, while the other only mentioned blog posts and new plugins - after i rebooted it, no less.

So i updated the first one, from 1.4.2 to 1.5.3, and updated all the plugins it told me about, and manually forced an update check on the other. That other one still claimed "Everything is up-to-date"! So i rebooted it again, forced the update check again, but still: nothing. The logs also showed no error or anything else besides that the check was ran, and nothing new was found.

So i started to write this very comment here, and while checking which version exactly the not-updating Pi was running, there it was: the notification that updates are available.

Not sure why it's such a hit-or-miss, but it does seem as if forcing the update check does not really do a good job? ;-)

cp2004, 2021-01-22:
If you look at #3777, there is sometimes an issue with Github's rate limit that is 60 requests per hour. And this is per check, so one request per plugin + core OctoPrint. Force checking for updates on both instances could have easily used up all of this, so you have to wait for an hour before you can do any more updates.

The solution in #3777 is to specify a github token, so that it can authenticate and have a limit of 5000 requests per hour, introduced in 1.5.0. If you are using 1.5.x, then you can try that solution otherwise to update to 1.5.x you will have to wait for the update notification to show up when it can.

jangrewe, 2021-01-22:
Well, here's a FunFact™:
```
pi@prusabox:~ $ grep github .octoprint/config.yaml
            github: ad587e144110cb1f...
```
And yes, that very same token is on the other Pi, where it works fine! But still, on this Pi, everything's up-to-date pretty much the very second i click the button.

Anyways, i'm now on 1.5.3, but i don't trust that update function anymore...

bolsoncerrado, 2021-01-22:
Just for the record this behavior was already happening when I reported this back in 1.4.0 (OP)

foosel, 2021-01-22:
@jangrewe `octoprint.log` and `plugin_softwareupdate_console.log` please.