OctoPrint/OctoPrint #3540: BufferError: Existing exports of data: object cannot be re-sized

FormerLurker, 2020-04-27:
#### What were you doing?
Using a callback from a worker thread that calls ```self._plugin_manager.send_plugin_message```

1. Create a worker thread with a callback to a plugin that calls self._plugin_manager.send_plugin_message
2.  Run the call over and over for a time (it succeeds for a while).
3.  Eventually you will see the following exception:
2020-04-27 10:45:57,455 - tornado.application - ERROR - Exception in callback functools.partial(<function wrap.<locals>.null_wrapper at 0x000001391CCFB280>)
Traceback (most recent call last):
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\ioloop.py", line 605, in _run_callback
    ret = callback()
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\src\octoprint\vendor\sockjs\tornado\transports\websocket.py", line 97, in <lambda>
    self.server.io_loop.add_callback(lambda: self.send_pack(message, binary))
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\src\octoprint\server\util\sockjs.py", line 46, in send_pack
    return orig_send_pack(*args, **kwargs)
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\src\octoprint\vendor\sockjs\tornado\transports\websocket.py", line 92, in send_pack
    self.write_message(message, binary)
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\websocket.py", line 252, in write_message
    return self.ws_connection.write_message(message, binary=binary)
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\websocket.py", line 793, in write_message
    return self._write_frame(True, opcode, message, flags=flags)
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\websocket.py", line 776, in _write_frame
    return self.stream.write(frame)
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\iostream.py", line 406, in write
    self._handle_write()
  File "C:\Users\Brad\source\repos\ArcWelderPlugin\OctoPrint\venv3\lib\site-packages\tornado\iostream.py", line 872, in _handle_write
    del self._write_buffer[:self._write_buffer_pos]
BufferError: Existing exports of data: object cannot be re-sized

For an easy way to see the error, follow these steps:

1.  Install the ArcWelder plugin from this commit url:
```https://github.com/FormerLurker/ArcWelderPlugin/archive/f6ec5b0ac7d4dddceadc2edd6eecd3a310fde1ea.zip```
2. Drag a small gcode file into local storage and wait for Arc Welder to process it.
3. Repeat step 2 a few times (usually happens on the third or fourth attempt, regardless of the file size) and see the error.  It prevents all notifications from reaching the client.

Also, I checked octoprint\vendor\sockjs\tornado\transports\websocket.py, and these packets that cause these buffer errors are sent using add_callback, which is supposed to be thread safe, so I'm not sure what modifications could be made there.

I was able to fix the error by modifying tornado\iostream.py.  On line 871 you will find this:

```
                if self._write_buffer_pos > self._write_buffer_size:
                    del self._write_buffer[:self._write_buffer_pos]
```
Changing that to this
```
                if self._write_buffer_pos > self._write_buffer_size:
                    try:
                        del self._write_buffer[:self._write_buffer_pos]
                    except BufferError:
                        # can't resize the buffer the way we'd like, just recreate it
                        self._write_buffer = bytearray(self._write_buffer_size)
```
fixed the issue.  However, in my testing I've only ever seen ```self._write_buffer_size`` = 0, so using that parameter may be unnecessary.  I don't notice any performance issues using the above code, but I'm using a powerful desktop, so I'm not sure if the del operation is faster or slower than recreating the bytearray.  I'm also not certain if there are unintended side-effects to using the above patch.

#### What did you expect to happen?
The plugin messages should have reached the client.

#### What happened instead?
I received the BufferError shown above.

#### Did the same happen when running OctoPrint in safe mode?
NA, this is using a plugin, but it is the only plugin installed.  I get the same messages in Octolapse sometimes. 

#### Version of OctoPrint
1.4.0

#### Operating System running OctoPrint
Windows 10

#### Printer model & used firmware incl. version
NA, running from virtual printer

#### Browser and version of browser, operating system running browser
NA, the error happens server side in tornado\iostream.py

#### Link to octoprint.log
https://gist.github.com/FormerLurker/ba4cf710e34122fb02c189ec63e5b7b1

#### Link to contents of terminal tab or serial.log
NA, no gcodes are being sent

#### Link to contents of Javascript console in the browser
NA, no client errors

#### Screenshot(s)/video(s) showing the problem:
NA, the error is shown only in the console/log

I have read the FAQ.

FormerLurker, 2020-04-27:
Edit:  I've determined that this only seems to happen when using Python3, though I will keep testing.

foosel, 2020-06-05:
Can you check if you can still reproduce this on the current `maintenance` branch? I fixed a bunch of stuff in sockjs-tornado that might possibly have an effect on this.

FormerLurker, 2020-06-05:
Will do!  Thanks for taking a look!

FormerLurker, 2020-06-08:
I will be testing this very soon, most likely this week.  Sorry to keep you waiting, but I suddenly have more free time :)

foosel, 2020-06-23:
@FormerLurker Any news here?

FormerLurker, 2020-06-23:
Sorry, no, nothing yet :(

Right after my last commend my free time was nuked from orbit.  I'm in a recursive loop of increasing task depth ATM.  You know how that goes.  I will try REALLY hard to get this tested over the weekend.  Please bug me about it constantly, especially since I opened this issue.

foosel, 2020-07-17:
Consider this me bugging you ;)

FormerLurker, 2020-07-17:
I am camping today, buy have Monday off.  Will double back then.  However, I can say it has been working so far.  I will put it through a stress test for complete confirmation.  Sorry about my recent absence from the scene.  I needed a break.

foosel, 2020-07-17:
No worries friend, if anyone understands having to take a break it's me. I was hoping you'd only see that on your return - take your time off and enjoy it!

FormerLurker, 2020-07-21:
Gina, it does appear to work as expected now, hooray!  In fact, it's working better now than ever, and the progress update is quite smooth.  

[This is the commit](https://github.com/OctoPrint/OctoPrint/commit/8b939d89a5b66e23638035a0cb315e7bcc400a5b) from the maintenance branch that I tested.  I have not had a chance to test Octolapse yet (it uses a similar progress update), but I think we can call it closed since I know it can work.

Thank you so much for your help with this!  CLOSED!

closed 2020-07-21 by FormerLurker