OctoPrint/OctoPrint #3541: [Potential Bug}: API/StorageInterface.list_files for files doesn't return all children as expected

jneilliii, 2020-04-27:
#### What were you doing?

While working on my PrusaSlicerThumbnails plugin I noticed that when using the internal module [self._file_manager.list_files()](https://docs.octoprint.org/en/master/modules/filemanager.html#octoprint.filemanager.storage.StorageInterface.list_files) folders containing sub-folder but no files are missing, and as a result the files stored deeper in the tree do not get returned. For example given the structure below the returned json does not include `File 1` or `Sub Folder 2`.

* Folder 1
  * Sub Folder 1
    * File 1
  * Sub Folder 2

#### What did you expect to happen?

I expected the json to have all folder and files returned if they are empty or not.

#### What happened instead?

json did not return all data

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.0

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 81.0.4044.122 (Official Build) (64-bit)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

jneilliii, 2020-04-27:
After further investigation, I think this is a larger issue relative to recursion. It seems that the recursion only goes one level deep for folder, so anything deeper than `Sub Folder 1` is missing.

foosel, 2020-05-06:
Hm... I don't seem to be able to reproduce this here... Can you share a minimal code example that shows how you are calling things when encountering this?

jneilliii, 2020-05-06:
It happens with the API so an easy way to reproduce is start with a blank uploads folder, create a sub-folder `Level 1`, open that folder and create a sub-folder `Level 2`, open that folder and create a sub-folder `Level 3`, then open `/api/files/local` and look at the returned json.  You will not have a `Level 3` anywhere to be found. Now upload a file to `Level 3` and request `/api/files/local` again and notice both `Level 3` and the file you just uploaded are also not returned in json.  Let me know if that's still not clear or doesn't work the same for you in 1.4.0.

foosel, 2020-05-06:
You will if you tell it to actually do a recursive listing though. If you do not provide `recursive=true` you get the result you describe, that is correct however. OctoPrint will only go one level deep if you don't provide the `recursive` flag. For directories it will also look at the direct children of the directory - though I can't remember right now why it does that, I'll have to dig through the associated commits to figure that out.

Just as an example, JSON output without `recursive`:

``` json
{
  "files": [
    {
      "children": [
        {
          "display": "Level 2", 
          "name": "Level_2", 
          "origin": "local", 
          "path": "Level_1/Level_2", 
          "refs": {
            "resource": "http://localhost:5000/api/files/local/Level_1/Level_2"
          }, 
          "type": "folder", 
          "typePath": [
            "folder"
          ]
        }
      ], 
      "display": "Level 1", 
      "name": "Level_1", 
      "origin": "local", 
      "path": "Level_1", 
      "refs": {
        "resource": "http://localhost:5000/api/files/local/Level_1"
      }, 
      "type": "folder", 
      "typePath": [
        "folder"
      ]
    }
  ], 
  "free": 99272376320, 
  "total": 498945867776
}
```

and with `recursive`:

``` json
{
  "files": [
    {
      "children": [
        {
          "children": [
            {
              "children": [], 
              "display": "Level 3", 
              "name": "Level_3", 
              "origin": "local", 
              "path": "Level_1/Level_2/Level_3", 
              "refs": {
                "resource": "http://localhost:5000/api/files/local/Level_1/Level_2/Level_3"
              }, 
              "size": 0, 
              "type": "folder", 
              "typePath": [
                "folder"
              ]
            }
          ], 
          "display": "Level 2", 
          "name": "Level_2", 
          "origin": "local", 
          "path": "Level_1/Level_2", 
          "refs": {
            "resource": "http://localhost:5000/api/files/local/Level_1/Level_2"
          }, 
          "size": 0, 
          "type": "folder", 
          "typePath": [
            "folder"
          ]
        }
      ], 
      "display": "Level 1", 
      "name": "Level_1", 
      "origin": "local", 
      "path": "Level_1", 
      "refs": {
        "resource": "http://localhost:5000/api/files/local/Level_1"
      }, 
      "size": 0, 
      "type": "folder", 
      "typePath": [
        "folder"
      ]
    }
  ], 
  "free": 99271213056, 
  "total": 498945867776
}
```

jneilliii, 2020-05-06:
Then a documentation issue...recursive should be True by default?

```
List all files in storage starting at path. If recursive is set to True (the default), also dives into subfolders.
```

foosel, 2020-05-06:
Huh, now that's a good point.

Considering that there's this line in the API endpoint:

``` python
recursive = request.values.get("recursive", "false") in valid_boolean_trues
```

that's definitely a doc bug. Going to fix that.

foosel, 2020-05-06:
After some minor confusion I just realized that you were referring to the docs for `StorageInterface` above, where `recursive` actually is `True` by default. The default value for the internal module does not reflect the default value on the API endpoint. They are different, and documented accordingly. Here's the docs for the API endpoints in question (https://docs.octoprint.org/en/master/api/files.html#retrieve-all-files and https://docs.octoprint.org/en/master/api/files.html#get--api-files-(string-location)):

[image]

[image]

If you do not provide `recursive` to the REST API, it will set it to `False` on its call to `FileManager.list_files`. If you call `FileManager.list_files` internally however, `recursive` will default to `True`.

The reason for that is that when calling the internal module you usually want all files, whereas there are valid reasons to limit the amount of returned data on the web API.

closed 2020-05-06 by foosel