OctoPrint/OctoPrint #3542: [Request] Add default state for terminal tab filter settings

Sembazuru, 2020-04-28:
**Is your feature request related to a problem? Please describe.**
I almost never want to have the temperature messages not filtered. When I load OctoPrint in a browser for the first time (or force a refresh) when I first go to the terminal tab I have to re-enable the temperature message filter. The setting seems to be persistent as long as I don't close or force a refresh on the window/tab that is showing OctoPrint.

**Describe the solution you'd like**
I'd like to see in the terminal filter settings a checkbox for each configured filter that has the default setting of the filter, on or off.

This is a low priority request, but hopefully it is easy enough to be handled quickly as low-hanging-fruit. ;-)
With my luck, this "simple" request can only be reasonably done by refactoring 75% of the code base...

jneilliii, 2020-04-28:
There's a plugin for this (shameless plug, it's mine).

https://plugins.octoprint.org/plugins/active_filters_extended/

Sembazuru, 2020-05-05:
> There's a plugin for this (shameless plug, it's mine).
> 
> https://plugins.octoprint.org/plugins/active_filters_extended/

I've installed this plugin, and it works as advertised. Thanx.

foosel, 2020-05-07:
Closing then since there's a plugin.

closed 2020-05-07 by foosel