OctoPrint/OctoPrint #3545: [1.4.0] error while calling plugin enclosure; KeyError: 'file'

tedder, 2020-04-29:
Seen in [a random log entry on reddit](https://www.reddit.com/r/octoprint/comments/g9oenz/any_solution_to_stutters_ender_3_with_abl_running), right before a print completes:
```
2020-04-29 01:31:40,691 - octoprint.plugin - ERROR - Error while calling plugin enclosure
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1550, in on_event
    file_name = os.path.basename(payload["file"])
KeyError: 'file'
```

It doesn't have a file? Interesting. Obviously no repro, but it should be pretty easy to defend against.

tedder, 2020-04-29:
another few cases, farther back (in time) in the log. It's always followed by plugins.excluderegion but I think that's a red herring.
```
2020-04-28 20:34:44,366 - octoprint.plugin - ERROR - Error while calling plugin enclosure
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1550, in on_event
    file_name = os.path.basename(payload["file"])
KeyError: 'file'

2020-04-28 20:50:02,511 - octoprint.plugin - ERROR - Error while calling plugin enclosure
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1550, in on_event
    file_name = os.path.basename(payload["file"])
KeyError: 'file'
```

jneilliii, 2020-04-29:
Interestingly enough Ted, this is documented. file was deprecated in the past, and flagged to be removed in 1.4.0. Situatio, these are plugin issues not octoprint issues. 

https://docs.octoprint.org/en/master/events/index.html#file-handling

foosel, 2020-04-29:
Yep, hence closing this.

closed 2020-04-29 by foosel