OctoPrint/OctoPrint #3546: [1.4.0] comm layer: too many values to unpack

tedder, 2020-04-29:
Seen in [a random log entry on reddit](https://www.reddit.com/r/octoprint/comments/g9oenz/any_solution_to_stutters_ender_3_with_abl_running), right before a cancellation completes:
```
2020-04-28 21:21:41,361 - octoprint.util.comm - ERROR - Error while processing hook dashboard for phase queued and command G1 X0 Y{machine_depth}:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3332, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 180, in process_gcode
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 180, in <genexpr>
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
ValueError: too many values to unpack
```

Obviously no repro, the {machine_depth} looks suspect to me, but the generator must be getting out of hand for some reason.

tedder, 2020-04-29:
here it is again, same log entry, actually scrolling back (in time).
```
2020-04-28 21:00:51,396 - octoprint.plugins.tracking - INFO - Sent tracking event print_cancelled, payload: {'origin': u'local', u'throttled_mask': 327680, u'elapsed': 470, 'file': 'ad97b2d42b760f9a20532e380113a5b502f73517', u'throttled_now': False, u'throttled_past': True}
2020-04-28 21:00:52,031 - octoprint.util.comm - ERROR - Error while processing hook dashboard for phase queued and command G1 X0 Y{machine_depth}:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3332, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 180, in process_gcode
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 180, in <genexpr>
    CmdDict = dict ((x,float(y)) for d,x,y in (re.split('([A-Z])', i) for i in cmd.upper().split()))
ValueError: too many values to unpack
```

tedder, 2020-04-29:
edited to include the template. are you happy now, bot?

kantlivelong, 2020-04-29:
@tedder The stack trace mentions octoprint_dashboard. Is this reproducible with 3rd party plugins disabled? If not then it might be worth opening this with the plugin developer.

tedder, 2020-04-29:
That's true about the plugin. I'm reporting it third-hand. I'll go over there and open it and close here.

tedder, 2020-04-29:
actually before I close, @foosel does the stats dash show a disproportional number of errors from this plugin? (can't remember if that reporting is enabled)

foosel, 2020-04-30:
I had to take out third party plugin errors from the error tracking completely since they were drowning out core issues and eating up all my events, so I don't know.

foosel, 2020-05-06:
Third party plugin bug, closing.

closed 2020-05-06 by foosel