OctoPrint/OctoPrint #3547: Read-only User Account Has Normal User Privileges (It Includes User Group)

rshartog, 2020-04-30:
This bug description was post to the OctoPrint community forum by user a-aron: [https://community.octoprint.org/t/read-only-users-isnt-working/17541]. I am filing a bug report here on GitHub corresponding to that forum report. The lines prefixed with [SH] are comments I have added to a-arons original forum post.

**What is the problem?**
With the new Access Control, Read-Only users cannot be created. This is because every user created is forced to have the User's group associated with the account. This overrides all the settings in Read-Only users.

**What did you already try to solve it?**
Tried to create a normal user and a read-only user, unchecked the default option for the User's group as well.

[SH] I also tried splitting the user group into a min_user and user where min_user has only minimal privileges. My intention was to remove the user group from the read-only user and add the min_user group. I made the min_user group the default and the user group not the default and recreated the read-only user account. In the end; however, I was unable to prevent the new read-only user account from including the user group which gave the read-only account normal user privilege (including print and cancel abilities).

[SH] I tested the read-only user account and confirmed that I could start and cancel a job on the printer.

**Logs**
Screen shots of the errors. Logs have no information regarding this issue.
[image]
[image]
[image]

**Additional information about your setup**
Octoprint 1.4.0, OctoPi 0.17.0, Raspberry Pi 4 with Raspbian, Prusa i3 MK3S 3.8.1, Brave browser on Windows 10, also Safari on MacbookPro.

[SH] Octoprint 1.4.0, Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2, Prusa MK2.5S, Safari 13.1 on macOS Catalina 10.15.4.

foosel, 2020-05-05:
Should be solved by the above commit, ready for 1.4.1. There's still an issue with group storage to be tackled (see TODO comment in commit note), that will be looked into separately however.

SebDominguez, 2020-06-08:
I'm still having the same problem, all new users are forced to be in he "users" group. I've created a "CamViewerOnly" group with can only see the camera stream, but since they are automatically opt in the Users group they can have full access to normal logged users...

foosel, 2020-06-08:
As mentioned in the above note, a fix is ready for the upcoming 1.4.1 release. It has *not* been merged back to 1.4.0 (not critical enough for that). If you need this now, switch to running the `maintenance` branch, otherwise switch to the Maintenance RCs release channel and you'll get an update notification once the first RC for 1.4.1 is out ("when it's done")

commit 8330a1d0 (2020-05-05): 🐛 Fix readonly group behaviour

User group is no longer enforced on all users. It is pre-checked
for new users (unless made non default) but can be disabled. That
makes it possible to have users without groups (and thus no
permissions) so beware!

TODO: Issue with default=False being written to groups.yaml,
causing the users group to not be enabled by default. Will be
solved in another commit.

Closes #3547
closed 2020-08-04 by foosel