OctoPrint/OctoPrint #3548: [Request] Clean hotend / change filament command

rlaferla, 2020-04-30:
Cura needs to be more user-friendly.  Currently, to change filament to a different color, you have to heat up the hot end and extrude the old color.  There should be a command in the UI to do this.  Perhaps, each printer has a different temp setting for it, extrude length/time, etc...

jneilliii, 2020-04-30:
This is exactly what [Custom Controls](https://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls) are for. If not comfortable with editing config.yaml manually there is a plugin called [Custom Control Editor](https://plugins.octoprint.org/plugins/customControl/) that will allow you to add the buttons you need.

rlaferla, 2020-04-30:
It's not that you can't get this functionality.  It's that it should be easy and accessible without any configuration by the average user.  OctoPrint is too complex for the average user and is geared for advanced users.  There's no reason why the UI can't accommodate both types of users.

foosel, 2020-04-30:
There is. Someone needs to maintain all this. That someone is mostly me.

Help is welcome. Especially in the form of an externally maintained plugin that does this.

BeatSlayer, 2020-04-30:
There is an existing plugin for that: [Change Filament](https://plugins.octoprint.org/plugins/Change_Filament/)

foosel, 2020-04-30:
Even better, closing.

closed 2020-04-30 by foosel