OctoPrint/OctoPrint #3552: Plugin-manager pip error: TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'

chmarr, 2020-05-01:
#### What were you doing?

Brand new installation of OctoPi/OctoPrint. Viewed the "Setup / Plugin Manager" page.

I.e.: clicked the setup icon, clicked "Plugin Manager".

#### What did you expect to happen?

Something other than an error, and the ability to install plugins.

#### What happened instead?

The following banner message appeared:

```
The pip command could not be found or does not work correctly for this installation of OctoPrint - please consult the log file for details and if necessary configure it manually. No installation and uninstallation of plugin packages is possible while pip is unavailable.
```

#### Did the same happen when running OctoPrint in safe mode?

Did not try. There are no plugins installed at this time, as it is a brand new installation.

#### Version of OctoPrint

1.3.12

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Ender 3 Pro. Upgraded mainboard.

#### Browser and version of browser, operating system running browser

MacOS 10.15.4, Safari 13.1

#### Link to octoprint.log

https://gist.github.com/chmarr/9a06bacb27e2dee005723b80adafb325

The relevant lines in the log:

```
2020-05-01 15:29:35,638 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 280, in on_api_get
    if refresh_repository or not self._is_repository_cache_valid():
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 797, in _is_repository_cache_valid
    return mtime + self._repository_cache_ttl >= time.time() > mtime
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
2020-05-01 15:29:35,695 - tornado.access - ERROR - 500 GET /api/plugin/pluginmanager (fe80::884:9a2c:57f1:8d5a) 130.85ms
```

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

chmarr, 2020-05-01:
Update: A restart of the OctoPrint server cleared the message. I suspect that the case where fetching of the plugin data had not been executed or had never worked is not catered for. In my case, a "poll" of the data was re-triggered by the restart.

kantlivelong, 2020-05-01:
@chmarr Is this still considered an issue or can this be closed?

foosel, 2020-05-05:
It's also something that's fixed in 1.4.0, please update ;)

closed 2020-05-05 by foosel