OctoPrint/OctoPrint #3560: [Request]

yet-another-average-joe, 2020-05-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

When a plugin edits config.yaml, a backup file is created : config.backup. But one backup is not enough in some circumstances

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Maintaining a few backup files :  config.backup01,  config.backup02, and so on ; this very common in the software industry ; for example, Eagle CAD does it.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Doing backups by hand. Using a SSH console, or doing it from a computer after .octoprint is shared with samba.

**Additional context**
Add any other context or screenshots about the feature request here.

none

tedder, 2020-05-09:
Please edit the subject to describe your feature request.

If you are concerned about backups, why not use the backup UI for the whole system?

[image]

yet-another-average-joe, 2020-05-09:
I just learn now about the backup/restore ! :))) I never payed attention, probably because I've always been used to image my computers system disk : with OctoPi, I just put the micro SD card in another Pi, shrink partitions, and image the whole card on a regular basis ; or save individual files on a PC through network shares. I've been doing this for the last 2 years on OctoPi !

Now, I will also use this feature, it's less time consuming !

Long story short, this FR was an idea I had after I read this : https://github.com/Salandora/octoprint-customControl/issues/32#issuecomment-626177415 ; it's only about some basic auto backup or undo queue, just in case a plugin does something wrong with config.yaml. (never had such an issue, but some did)

tedder, 2020-05-09:
Cool. Yeah, I could see the desire for an autobackup type feature, but it seems like a lot of work for little gain.

closed 2020-05-09 by tedder