OctoPrint/OctoPrint #3562: Refactor comm.py and other large files to improve maintainability

DanielJoyce, 2020-05-09:
I'm already working on a PR to break apart comm.py, but I need to know if this needs to land against master, devel, or maint? Its possible to refactor it and have all tests pass.

I know there is ongoing work on Comm Next Generation, but breaking the existing file apart right now would provide a lot of benefits. It would make it easier to understand and browser. 

There are other giant files as well that could stand to be refactored. I could also issue a small set of PRs that just break things apart piecemeal.

jneilliii, 2020-05-09:
Not sure these changes will be considered since @foosel is already re-writing the the entire comm layer already.

DanielJoyce, 2020-05-11:
The rewrite is ongoing for a long time. Anything that makes support of the current code base easier should be welcome.

DanielJoyce, 2020-05-11:
Also, if I clean it up, I have some plans for making serial port handling more robust.

foosel, 2020-05-12:
Never `master`. Refactoring is an improvement, so `maintenance`. I'm really unsure about this though. As long as it's really only breaking it apart, as in, refactoring it from one to multiple files, it's something I'd be willing to do, but in general ANYTHING that touches `comm.py` at this point just makes the bloody rewrite take longer because I need to port it over (or at the very least decide whether I need to port it over or not). In a nutshell, this might just delay things further and make them more difficult, and I already struggle with everything on my task list as is.

Long story short, pure refactoring to break into several pieces, maybe ok (needs proper review first which I can't do right now, and I'm honestly not sure it will actually improve things as it might introduce hard to debug issues that will then cause even more maintenance overhead). Anything else touching the comm layer will not be merged, it will only add even more TODOs to an already never ending list.

DanielJoyce, 2020-05-12:
yes, the plan would be pure refactoring because right now its a giant god package.

foosel, 2020-05-15:
As I said, happy to review a PR against `maintenance` then, minus all the unnecessary cosmetic whitespace changes that you put into #3561 as noted there.

closed 2020-08-04 by foosel