OctoPrint/OctoPrint #3564: Octopi.txt not working for camera on new install

jmlumpkin, 2020-05-10:
#### What were you doing?
I was setting up a new printer, and ran a few updates. I then realized my pi cam was defaulting back to 640x480. I have always had "-x 1280 -y 720 -fps 10", and this worked fine. This was even in my octopi.txt file, but was not being read, after many restarts, etc. 

I then did a complete reinstall of octopi 0.17, made the edit to octopi.txt, and still it is defaulting to 640x480. If I look at the output of 'ps', it notes `./mjpg_streamer   -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0`

Also, I noticed in webcamd, that it is looking for a config file folder, like /boot/octopi.conf.d/

1. install new copy of octopi 0.17 on a Pi 3B+
2. Edit Octopi.txt to use `camera_raspi_options="-x 1280 -y 720 -fps 10"`
3. Install and start raspberry pi
4. Camera defaults to 640x480

#### What did you expect to happen?
I expected to see a different view on my camera

#### What happened instead?
Showed in default 640x480. 

#### Did the same happen when running OctoPrint in safe mode?
Brand new install with default config. 

#### Version of OctoPrint
1.3.12

#### Operating System running OctoPrint
Octopi 0.17

#### Printer model & used firmware incl. version
Ender 3, but never even set it up yet. 

#### Browser and version of browser, operating system running browser
Up to date Chrome on Windows 10. 

I have read the FAQ.

jmlumpkin, 2020-05-10:
Also, updated to 1.4 and still have the same issue.

jmlumpkin, 2020-05-10:
After testing, by having camera set to 'auto', this is where it failed. If I changed to just 'raspi', it worked. The error that I could find was:

```
Found video devices:
/dev/video0
/dev/video10
/dev/video11
/dev/video12
raspi
config file='/boot/octopi.txt':USB device was not set in options, start MJPG-streamer with the first found video device: /dev/video0
<13>May 10 22:33:10 pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1196444237 res: 640x480
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
Done bring up all configured video device

Goodbye...
```

Auto seemed to work just fine before, why would this start acting up, even on a new install?

fieldOfView, 2020-05-11:
Since the camera is not handled by OctoPrint, but by an external component in your OctoPi installation, this is not an OctoPrint issue. You may want to open an issue over at the OctoPi issue queue (https://github.com/guysoft/OctoPi/issues), or seek help in the OctoPrint community (https://community.octoprint.org/).

closed 2020-05-11 by fieldOfView