OctoPrint/OctoPrint #3568: GPX Plugin: 2.6.3 update to 2.6.5 wrong download location?

wolfbiteaus, 2020-05-14:
#### What were you doing?
upgrading GPX Plugin: 2.6.3 update to 2.6.5

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?
Updating, please wait./home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gzERROR: HTTP error 404 while getting https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gzERROR: Could not install requirement https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz because of error 404 Client Error: Not Found for url: https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gzERROR: Could not install requirement https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz because of HTTP error 404 Client Error: Not Found for url: https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz for URL https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz

Could not install requirement 
https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.5/OctoPrint-GPX.tar.gz because of HTTP error 404
*** Wrong download address?? ***
https://github.com/markwal/OctoPrint-GPX/releases/download gives 404
Manually checking location and getting the address from the file gives
https://github.com/markwal/OctoPrint-GPX/archive/2.6.5.tar.gz

(I dont know where to edit and see if it fixes)

#### Version of OctoPrint
OctoPrint 1.4.0 running on OctoPi 0.17.0

#### Operating System running OctoPrint
Octopi

I have read the FAQ.

foosel, 2020-05-14:
Please report this with the GPX plugin repository, there's absolutely nothing that I can do about that, @markwal however might be able to solve it.

closed 2020-05-14 by foosel

markwal, 2020-05-14:
Oops sorry about that.  Since the plugin has a submodule the normal github archive doesn't include all the sources, so I have to manually include one and I forgot.  :-(