OctoPrint/OctoPrint #3569: [Request] Octoprint makes recent feedrate changes available via API

thegarbz, 2020-05-15:
Octoprint allows users to change feedrate and flowrate of the print in progress. This gets sent to the printer. There is no way to query the current operation of the printer and that is fine. The Octoprint dashboard displays the previous feedrate change. 

The problem comes with other interfaces (OctoDash / OctoScreen) no longer reflect the feedrate which was sent to the printer as this is not made available via the API according to the developer. This is frustrating if the feedrate is changed e.g. on the web interface vs an LCD display. 

It would be good if OctoPrint can report via API the last feedrate / flowrate change it sent to the printer so that the Control tab on the web interface and any other interfaces to Octoprint can be in sync. 

An alternative currently being investigated in Octodash is for each interface to remember its own previous setting. This lets all interfaces get out of sync very quickly. The solution would not be ideal since it is still possible to override this on the printer itself, but at least being internally consistent in Octoprint and plugins would be a good start.

foosel, 2020-05-15:
I'm a bit worried about that tbh. As soon as the user modifies these settings (and fan rate, selected tool etc for that matter), that OctoPrint API would lie. And what good is an API that you cannot rely on to report the correct state? 

I'd rather leave that in the sorry state it is in to maybe motivate someone to implement query possibilities into firmware (which would be the proper solution for this whole mess), instead of adding yet another simulation of functionality that isn't really technically possibility and which then gets demanded to get fixed in regular intervals.

In fact, the feed and flow rate controls in `maintenance`/1.4.1 have been slightly reworked and now also have this note placed on them:

[image]

thegarbz, 2020-05-15:
I agree, an API should not lie. But then it wouldn't lie, it would only assure that the entire Octoprint system remains internally consistent. The API wouldn't lie anymore than me opening up Octoprint.local/ clicking on control and seeing that it was set to 75% because I did that an hour ago. On the flip side the LCD is lying to me because it still shows that the feedrate is 100%.

I agree the best way to solve this would be for Marlin or other firmwares to implement the ability to query or for the printer to send an update if its set locally. 

But surely at the least resolving the internal inconsistency is better than the current state. I get it APIs are for external access to Octoprint, but those same APIs are used to render some interfaces, so right now not only does the value on the UI not reflect the current setting, but the value on the UI in one place doesn't even reflect the value of the UI in another place, and that is entirely within the control of the API.

foosel, 2020-05-15:
There is no consistent state because there is no state. If anything OctoPrint should probably not show you whatever you set earlier either and just delete the value there immediately on send and/or focus loss. As things are, it's a write-only interface. It doesn't make sense to simulate it being anything else and fake consistency where there simply isn't any. That's just going to cause confusion further down the road for users.

In fact, this argument here has convinced me to remove the current misleading value display completely and make sure it's completely nuked and empty unless the user has just typed something in in preparation for sending - apparently it's only motivating clients to imitate the display behaviour for values that are ephemeral. The only truth lies with the firmware, and it won't share.

thegarbz, 2020-05-15:
Perfect solution actually. Not displaying the value, or setting it to --- every time tab focus is lost provides just the consistency that is needed in the software and removes all doubt. I will raise the same request to the Octodash developer. 

(the whole reason this came up is because Octodash only sent changed values, and by always defaulting to 100 meant that this morning could not set the printer back to 100, only 99 or 101, so I went down this rabbit hole :) ).

Cheers

foosel, 2020-05-15:
Just for the record, the commit above implements that as outline and it now looks like this:

[image]

Ready for 1.4.1.

thegarbz, 2020-05-15:
Perfect. Clear, and will hopefully prevent the kind of confusion I had in the future. Thanks for your great work.

closed 2020-08-04 by foosel

DyGyg, 2020-08-12:
hi! is it possible to switch back to old style controls with sliders?
i understand that there is no such info from printer and its only "setters". 
and for me it's more comfortable to set it with sliders

foosel, 2020-08-12:
A slider makes absolutely no sense with a write only value, there's nothing I could set it to after firing off the set command that wouldn't be interpreted once again as "this is the current value" and restart the whole confusion and discussions.

This is what works given the protocol limitations, and until there's a universally implemented and *detectable* way to query the values from the printer, it'll stay this way. If you want a slider, you can add an additional custom control, to see https://docs.octoprint.org/en/master/features/custom_controls.html

DyGyg, 2020-08-12:
thanks for quick answer
i've pulled 5ca20f7 and now it's much better to understand the logic of this control - if the field is empty - nothing to set, so "Set" button is disabled. i think its good idea and i will get used to it)

commit b0d3c4a5 (2020-05-15): 🚸 Reflect on UI that feed/flow rate are ephemeral

We can't query the setting from the firmware, and we also can't reliably
track it through OctoPrint as it could be modified through GCODE
triggered through a file on the SD card and/or interactions with the
controller. Hence we now properly reflect that this value is write only
by emptying it 5s after focus loss (with a little opacity pulse to warn
about that) and immediately on sending.

That should hopefully make it a bit clearer that this value cannot be
trusted and will never reflect the firmware value unless query commands
or status repor [... cut]