OctoPrint/OctoPrint #3570: Missing documentation for Octoprint 1.3.11

LuckyX182, 2020-05-15:
I'm going to be quick here with description, there's documentation missing for OctoPrint 1.3.11 on docs.octoprint.org.

tedder, 2020-05-15:
Indeed this is true- and the full version of 1.4.0 isn't there, just 1.4.0rc5. 1.3.12 is there.

"bad"/missing
https://docs.octoprint.org/en/1.3.11/
https://docs.octoprint.org/en/1.4.0/

"good"
https://docs.octoprint.org/en/1.4.0rc5/
https://docs.octoprint.org/en/1.3.12/
https://docs.octoprint.org/en/1.3.10/

foosel, 2020-07-22:
Doc build has been revamped, docs for 1.3.12 and 1.4.0 are there, also for the current 1.4.1 RC and for `master`, `maintenance` and `devel`. It doesn't make sense to keep docs for all prior versions, so I'm going to do a rolling window here of n-1, n, n+1 during RC phases, and n-2, n-1, n outside of RC phases.

So: currently it's 1.3.12, 1.4.0, 1.4.1rc3, once 1.4.1 gets a stable release it'll be 1.3.12, 1.4.0, 1.4.1, once 1.4.2 hits RC it'll become 1.4.0, 1.4.1, 1.4.2rcX.

If you need older docs, the sources are in git for each and every release.

closed 2020-07-22 by foosel