OctoPrint/OctoPrint #3571: [Request] When uploading files to octoprint, ensure file mode is 0o660 so other programs that have octoprint group permissions can see them

DanielJoyce, 2020-05-15:
**Is your feature request related to a problem? Please describe.**
Files uploaded to octoprint are created with permission 600, or rw-------, this means members of the octoprint group can't see them. It's common when deploying octoprint on rpi or unix to give it its own user and group

In my case, I have created a user/group for klipper, and another pair for octoprint. With the octoklipper plugin installed, and klipper virtual-sd config set to point to /opt/octoprint/.octoprint/uploads, klipper can not see any the files uploaded to uploads, even though I added the octoprint group to the klipper user. That is because the files are created as 600, so only visible to the octoprint user, but no one in the octoprint group can see it.

**Describe the solution you'd like**
I would like it so that when uploading files to octoprint, the file permissions would be set to 660, thus facilitating sharing to other programs who are members of the octoprint group, but not the octoprint user.

**Describe alternatives you've considered**

Use group permissions to share access is the simplest and common solution on linux.

**Additional context**

Working on a PR

DanielJoyce, 2020-05-15:
Adding the following to the end of add_file of LocalFileStorage class fixes the problem for me.

``` python
                # touch the file to set last access and modification time to now
                os.utime(file_path, None)
                os.chmod(file_path, 0o660)
```

But this means every file added will have group permissions set. This might be okay, or might not.

We can't necessarily change the api, so perhaps passing it in via the metadata would be an option?

blalor, 2020-07-30:
I came here to report this issue, as well. I have OctoPrint writing to an NFS-mounted directory and the `600` mode is a problem for me, as well.  I wonder if the [Daemon](https://github.com/OctoPrint/OctoPrint/blob/e9b6d22932a9baa81668dab884fce4b2dbe384a7/src/octoprint/daemon.py#L48) class is responsible?  I'd prefer to have OctoPrint honor whatever is set by the launcher (systemd in my case), rather than an OctoPrint-specific setting.

foosel, 2020-07-31:
I just checked, and the only place OctoPrint sets a umask is in fact in the Daemon class (which you btw shouldn't be using if you are using systemd - in fact this whole construct is something that I'd rather remove at this point...), and that is only a umask of 0002, so only other gets denied, group should not be restricted by this. 

There are a bunch of files it will force to be written with 0600 (config files that contain secrets, user & group files - so intentionally and with security reasons), but everything else should stick to the default umask set on the process.

However, it isn't, and that is actually a bug. From the looks of it it might be caused by using the `tempfile` module in some places, e.g. for upload buffering, which will set quite restrictive permissions (which - for temporary files - is actually a good idea) which then on move to the final destination will be kept. Which leads to buggy behaviour.

blalor, 2020-07-31:
Ooh, `tempfile` makes a ton of sense.  Thank you for addressing this!

foosel, 2020-08-04:
Fix ready for 1.4.2

foosel, 2020-08-06:
Make that 1.4.3, 1.4.2 just got turned into a hotfix release.

foosel, 2020-08-10:
Actually make that 1.5.0 because I've decided to finally *fully* adhere to semver and maintenance releases will see a minor increase in the future instead of patch, patch will be used for hotfixes if needed.

blalor, 2020-08-10:
😂

I think that’s a great idea!

commit 89417891 (2020-07-31): 🐛 Set sensible permissions on (uploaded) files

chmod to 0o664 $ ~UMASK after moving to final destination.

Should fix #3571
closed 2020-11-30 by foosel