OctoPrint/OctoPrint #3572: No Permisson to upload files

CStiller, 2020-05-16:
I'm using OctoPrint 1.4.0 running on OctoPi 0.17.0. 
OctoPrint 1.3.12 was working.
After the update of Octoprint to 1.4 Prucaslicer(2.2.0-linux-x64) do not have the permission to upload files.

Error from Prucaslicer:

Error uploading to print host:
HTTP 403: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>403 Forbidden</title>
<h1>Forbidden</h1>
<p>You don't have the permission to access the requested resource. It is either read-protected or not readable by the server.</p>

Log from Octoprint:

2020-05-16 23:24:40,730 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.4
2020-05-16 23:24:40,991 - octoprint.server.util.flask - INFO - Passively logging in user _admin from ::ffff:192.168.0.4
2020-05-16 23:24:40,995 - octoprint.access.users - INFO - Logged in user: _admin
2020-05-16 23:24:41,087 - octoprint.server.util.sockjs - INFO - User _admin logged in on the socket from client ::ffff:192.168.0.4
2020-05-16 23:25:43,227 - tornado.access - WARNING - 403 POST /api/files/local (::ffff:192.168.0.4) 203.51ms
2020-05-16 23:26:04,869 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.4

This has clearly to do with OctoPrint handling  transmission differently in the new version. Does this belong here? - I don't know!! Anyway thank you for your time!

CStiller, 2020-05-17:
So again with "proper from" !!!

#### What were you doing?
Try to upload files from PrucaSlicer

1. Updating Octoprint to 1.4.0
2. Slice a Part with PrucaSlicer 2.2.0-linux-x64
3. Click upload to Printer Button

#### What did you expect to happen?
Usually it the G-code should be accessible on the web client

#### What happened instead?
PrucaSlicer Error:
`
Error uploading to print host:
HTTP 403: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>403 Forbidden</title>
<h1>Forbidden</h1>
<p>You don't have the permission to access the requested resource. It is either read-protected or not readable by the server.</p>
`

OctoPrint log:

`
2020-05-16 23:24:40,730 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.4
2020-05-16 23:24:40,991 - octoprint.server.util.flask - INFO - Passively logging in user _admin from ::ffff:192.168.0.4
2020-05-16 23:24:40,995 - octoprint.access.users - INFO - Logged in user: _admin
2020-05-16 23:24:41,087 - octoprint.server.util.sockjs - INFO - User _admin logged in on the socket from client ::ffff:192.168.0.4
2020-05-16 23:25:43,227 - tornado.access - WARNING - 403 POST /api/files/local (::ffff:192.168.0.4) 203.51ms
2020-05-16 23:26:04,869 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.4
`
#### Did the same happen when running OctoPrint in safe mode?
Yes!  Same in save mode!!

#### Version of OctoPrint
OctoPrint 1.4.0 

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
not relevant but Two Trees Sapphire Pro

#### Browser and version of browser, operating system running browser
Also not Imoprtant:
Firefox 76.0.1 (64-bit)
On Ubuntu Mate 18.04

#### Link to octoprint.log
Only truncated version accessible for now! See above! If full is requested I can provide it later.

#### Link to contents of terminal tab or serial.log
Not necessary!

#### Link to contents of Javascript console in the browser
Not necessary!

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

fieldOfView, 2020-05-18:
Did you enter a correct API key in the PrusaSlicer configuration?

foosel, 2020-05-18:
> If full is requested I can provide it later.

Hereby requested. Can't work without it.

chimcen, 2020-05-18:
@CStiller 
Why so rude?

CStiller, 2020-05-18:
Hey there!
thank you for all your replies! 

@chimcen  sorry for being rude. Didn't mean it.....
@foosel yes next time I will post the full log! Now I know how to do it right!
@fieldOfView The Api Key was new to me! It worked! 

I'm a bit ashamed of posting such a nooby question here in regards to the solution...
You are awesome! Great help - great Projekt!

Have a nice Day
CStiller

closed 2020-05-18 by CStiller

fieldOfView, 2020-05-18:
No problem. A good place for nooby questions is the community forum: https://community.octoprint.org