OctoPrint/OctoPrint #3573: [Request] UPS functionality/Safe cool down when UPS is low

wrpetersen, 2020-05-17:
**Is your feature request related to a problem? Please describe.**
I have fairly common power outages as my house is on a mountain in a forest. I have an uninterruptible power supply for my 3d printer but would like a feature to pause/cool down/ shutdown when the UPS is low.

**Describe the solution you'd like**
Many uninterruptible power supplies have a USB output to signal servers/PCs/NAS boxes etc when the battery is near empty so that the machine can shut down gracefully before it cuts out. It would be excellent if the octopi server could monitor a UPS as well and when the battery is at a specified level it will stop the print at a place that will be easy to resume, cut the heaters, and allow the hot end to cool before shutting down gracefully. For those like me with PSU controls, it would be cool if the power also cuts off to the printer to avoid surges on power restoration. It would be wise to be able to set that threshold to account for printer wattage/UPS size so that it will continue printing on short power outages so long that the battery doesn't drain to a certain level.

**Describe alternatives you've considered**
N/A 

**Additional context**
N/A

BeatSlayer, 2020-05-18:
Sounds like a good plugin candidate.

Andy-ABTec, 2020-05-19:
apcupsd provides the necessary hooks for APC UPS's and can be installed onto a Pi with apt, trouble is I'm no coder - if I was something similar would have been written a while ago...

jneilliii, 2020-05-19:
This was discussed on the community forum before, but I don't think the plugin ever got made.

wrpetersen, 2020-05-21:
Does the apcupsd support just APC or others? I believe that there is likely some base legacy protocol available as I have a NAS server that automatically recognizes both Cyber Power and APC brand UPSs through its USB port without any drivers (or maybe Netgear built some really good firmware with multiple brand specific drivers but is doubtful). I would think the plugin should support at least both of the two major competitors.

Andy-ABTec, 2020-05-21:
It only supports APC UPS's to my knowledge - see http://apcupsd.com/. Just out of interest what model is your Netgear NAS? - I have a couple of 8Tb 1100's and they only support APC UPS's...

wrpetersen, 2020-05-21:
Mine is just a home level ReadyNas Duo V2 and I could swear it was plugged
into my old cyber power UPS but now I am wondering if I got that mixed up.

timur-tabi, 2020-12-08:
There is an apcupsd plugin for octoprint: https://github.com/OutsourcedGuru/OctoPrint-APCUPC

However, development on it has apparently ceased and it's no longer compatible.  I have apcupsd running on my OctoPi connected to my UPS, but it's only configured to do a proper shutdown if the battery gets low.

I'm hoping that someone forks the repo and continues development.

jneilliii, 2020-12-08:
You should reach out in the forum. OutsourcedGuru is still around, just been busy with building an off grid home.

NotExpectedYet, 2021-01-05:
NUT might be a better alternative to apcupsd as it supports a wider variety of UPS' and I'm currently using it plugged into a Pi with two attached.

kantlivelong, 2021-11-03:
This exists:
https://plugins.octoprint.org/plugins/ups/

jneilliii, 2022-01-22:
Closing now that plugin is registered and available.

closed 2022-01-22 by jneilliii