OctoPrint/OctoPrint #3574: [Request] Integration of WebCamStream into Terminal Tab

chimcen, 2020-05-18:
**Is your feature request related to a problem? Please describe.**
When using terminal it would be awesome to see what happened on printer side.
Actually i've the switch to control tab to see whats happening after a terminal input.

**Describe the solution you'd like**
Could the webstream integrated into terminal tab (left or right beside the term window)?
It would also be cool if the webstream could be easily turned on/off in settings menu of octoprint for the terminal tab.

**Describe alternatives you've considered**
-

**Additional context**
-

BeatSlayer, 2020-05-18:
Give a try to: [WebcamSB](https://plugins.octoprint.org/plugins/webcamSB/)

chimcen, 2020-05-19:
This is great... its small - but enough for me...

closed 2020-05-19 by chimcen