OctoPrint/OctoPrint #3576: Make video stream password protected

purist69, 2020-05-20:
I've been exposing my octoprint server for a couple of years now over SSL with LetsEncrypt certs.  However, I just realized through mucking around that https://URL/webcam/?action=stream and http inside my network both expose the video stream without prompting for credential.

Would this not be a possible entry point for hackers to gain access to someones network?

jneilliii, 2020-05-20:
You should check the community forum for this as exposing octoprint to the public internet is frowned upon. I'm pretty sure there are posts relative to fixing this, search for haproxy.

fieldOfView, 2020-05-21:
Since the video stream is not provided by OctoPrint, but by an external component that comes with OctoPi, this is not something that can be fixed in OctoPrint. As @jneilliii suggests, please check the [OctoPrint community](https://community.octoprint.org) instead, and at the very least read this article: https://octoprint.org/blog/2018/09/03/safe-remote-access/

closed 2020-05-21 by fieldOfView

Moorviper, 2020-05-21:
https://github.com/jacksonliam/mjpg-streamer/issues
is the way to go...