OctoPrint/OctoPrint #3578: Raw websockets no longer work in maintenance branch (unreleased code)

gdombiak, 2020-05-26:
#### What were you doing?

1. Install OctoPrint from **maintenance branch** (after commit 5/6/20 **9e3cf58b**)
2. Connect to printer (virtual or physical is the same)
3. Open OctoPod (that uses the OctoPrint instance) or some client that uses **raw websockets**
4. See error

> 2020-05-26 03:55:41,223 - tornado.general - ERROR - RawWebSocket
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 63, in on_message
    self.session.on_message(message)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 27, in on_message
    self.conn.on_message(msg)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/server/util/sockjs.py", line 210, in on_message
    self._on_login(user)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/server/util/sockjs.py", line 448, in _on_login
    self._logger.info("User {} logged in on the socket from client {}".format(user.get_name(),
AttributeError: 'NoneType' object has no attribute 'get_name'

#### What did you expect to happen?

Have a working websocket (like it used to be before mentioned commit)

#### What happened instead?

Raw websocket do not work

#### Did the same happen when running OctoPrint in safe mode?

n/a

#### Version of OctoPrint

UNRELEASED CODE. Maintenance branch. Make sure to have >= commit https://github.com/OctoPrint/OctoPrint/commit/9e3cf58bc9941540f16a2fa55a8c666499a3ad55

#### Operating System running OctoPrint

N/A. Happens on raspbian and macOS

#### Printer model & used firmware incl. version

N/A. Virtual or physical

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

> 2020-05-26 03:55:41,223 - tornado.general - ERROR - RawWebSocket
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 63, in on_message
    self.session.on_message(message)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 27, in on_message
    self.conn.on_message(msg)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/server/util/sockjs.py", line 210, in on_message
    self._on_login(user)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/server/util/sockjs.py", line 448, in _on_login
    self._logger.info("User {} logged in on the socket from client {}".format(user.get_name(),
AttributeError: 'NoneType' object has no attribute 'get_name'

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

kantlivelong, 2020-05-26:
HAH I just ran into this myself and pushed PR #3579 which should fix this.

gdombiak, 2020-05-26:
LOVE it! That PR should fix it!

Thanks,
Gaston

foosel, 2020-06-04:
Now merged. 

@kantlivelong good job! Should also make things perform slightly better. One minor request: don't mark as solved until merged next time, I was a bit confused when I couldn't find the ticket in the "to fix" project column just now 😅

foosel, 2020-08-04:
That's actually fixed in 1.4.1 which was just released, just forgot to add this ticket to the milestone.

closed 2020-08-04 by foosel