OctoPrint/OctoPrint #3580: Are we missing the top half of alignment picture?

9Nazguls, 2020-05-27:
[image]

tedder, 2020-05-27:
wait, what? Where do you see this alignment picture?

9Nazguls, 2020-05-27:
http://192.168.0.59/#tab_plugin_mgsetup_maintenance-cleanup

9Nazguls, 2020-05-27:
Maintenance Index
Independent Dual Adjustments
Dual X Adjustments

9Nazguls, 2020-05-27:
Just noticed Dual Y Adjustments is missing the left half of the alignment picture.
I thought I followed the directions properly….

tedder, 2020-05-27:
I can't see your private Octoprint server, obviously. However, it appears to be a plugin of some sort. Do you know what plugin it is?

9Nazguls, 2020-05-27:
Just the standard maintenance functions

Maintenance Index
Independent Dual Adjustments
Dual X Adjustments
  -and-
Dual Y Adjustments

Deleted the following plugin:

MakerGear Setup (0.3.0.9)
TODO
 Homepage<https://github.com/MakerGear/MakerGear_OctoPrint_Setup>  MGDevTeam

Checking to see if that clears it.  There’re not many plugins installed since I don’t know what I’m doing.

Nope, now I don’t have the Maintenance Index at all….

Reloading….

tedder, 2020-05-27:
Yeah, that's a problem with [the MakerGear plugin](https://github.com/MakerGear/MakerGear_OctoPrint_Setup/). [You can actually see similar filenames in their plugin repo](https://github.com/MakerGear/MakerGear_OctoPrint_Setup/tree/master/octoprint_mgsetup/templates), which explains why I've never seen it before- I've never used that plugin. You should open an issue in that repo, it isn't an OctoPrint issue.

closed 2020-05-27 by tedder

9Nazguls, 2020-05-27:
Thanks Ted