OctoPrint/OctoPrint #3582: Announcement Icon not showing in Nav Bar

pedrocadiz13, 2020-05-29:
I've noticed recently that  the Announcements Icon is not showing in the Nav Bar

I'm logged in as Admin, and have tried enabling  permissions 

PLUGIN_ANNOUNCEMENTS_READ
PLUGIN_ANNOUNCEMENTS_MANAGE via Access control.

I think the issue is caused by the PLUGIN_ANNOUNCEMENTS_ANNOUNCEMENT permission not  being granted.

Have fixed locally by changing line 152 in __init__.py to:

			dict(type="navbar", template="announcements_navbar.jinja2", styles=["display: none"], data_bind="visible: loginState.hasPermission(access.permissions.PLUGIN_ANNOUNCEMENTS_READ)")

Although  this might not be the best way to fix.

Hope this helps.

foosel, 2020-05-29:
That's a known issue and already fixed for 1.4.1. But thank you for reporting!

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel