OctoPrint/OctoPrint #3586: [Request] A way to list and define the order of gcode processing plugins

corporategoth, 2020-05-31:
**Is your feature request related to a problem? Please describe.**
These days, there are multiple plugins that process gcode.  Both when it is received by Octoprint, and when it is being streamed to the printer.  However, there is no way to
a) See or specify the ORDER in which those plugins run (where the output of one becomes the input of the next)
or b) For any plugin to abort processing (prevent the gcode passing to the next plugin).

**Describe the solution you'd like**
I would like a simple config screen that shows me a list of which plugins have hooked into the process-gcode-on-upload hook and the process-gcode-as-streamed hook.   And to be able to drag plugins up or down in the order (or bypass them for that hook).

I would also like for a plugin to signal 'ABORT!' when processing a stream.  Though this is less of a priority if subsequent plugins just read the same file, from the beginning, after the previous plugin is done with it (the previous plugin can signal abort by simply deleting the file).

**Additional context**
This was inspired by things like the interaction between ArcWelder and things like DisplayLayerPlugin, or PrintTimeGenius.  ArcWelder is going to completely re-write the gcode, either into a new file or replacing the existing one.  There is NO reason to perform the time estimation, or put in layer markers until AFTER ArcWelder has generated a new file, because we just end up doing the same processing twice.

Additionally, other plugins could modify the gcode in similar ways, which could affect the time estimation from PrintTimeGenius, so I would wan to be able to say 'That is the last plugin that touches the gcode file when imported.'

I can imagine similar issues for streaming gcode to the printer, where gcode is being inline changed (for example by octoprint, or displaylayermanager) and ordering is important (eg. you could tell octoprint to look for M117 INDICATOR-Layer.* to trigger it's camera shot, but displaylayermanager is going to replace that indicator with a nice display (ie. 58% L58/80)).  So being able to specify this order would be helpful too.

jneilliii, 2020-05-31:
DisplayLayerProgress actually handles the order internally rather than calling the standard callback like everyone else, forcing it to run last. That is a workaround that other plugin authors could also incorporate. It's done [here](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/blob/9e54f97d278b27b185ada28e0c2dc9f7ffddccbc/octoprint_DisplayLayerProgress/__init__.py#L1035) in the on_after_startup callback. You could use a similar approach to force it to the beginning, but then you may still see ordering conflicts if multiple plugins are doing the same thing. Even with an ordered approach internally, how do you deal with multiple plugins using the number 0 for their order, default to identifier? I believe that's how other order related stuff is done in OctoPrint callbacks now.

corporategoth, 2020-05-31:
@jneilliii I was more thinking that I could specify the order myself in a config table, not that the plugin would define the order.

In other words, I am shown something like:

| + | Arc Welder |
| + | Display Layer Progress |
| + | Print Time Genius |

Based on the currently enabled plugins and which is hooking into the gcode processor, and I could just drag up or down.

Similarly for output to printer:

| + | Dashboard |
| + | Display Layer Plugin |
| + | Octolapse |

Where I could drag octolapse up, for example, from the settings menu.

kantlivelong, 2020-06-01:
Not a bad idea really. Also might be worth allowing plugin devs to specify a priority value so that manually changing the order isn't required as often.

foosel, 2020-06-01:
Y'all are aware of 

https://docs.octoprint.org/en/master/plugins/hooks.html#execution-order

and

https://docs.octoprint.org/en/master/plugins/mixins.html#execution-order

? That's been available for ages.

Now, those don't allow manual reordering, but frankly I also smell a TON of misconfiguration potential there that would cause hard to debug and this support expensive issues.

I've thought about allowing to define explicit before/after rules, but resolving those can become a nightmare.

kantlivelong, 2020-06-01:
@foosel huh I've never noticed that in the docs nor the code. TIL

jneilliii, 2020-06-01:
I knew of execution order, which is what I meant by "rather than calling the standard callback like everyone else" but pointed out how it can also be overwritten. That specific use case was because there is no option for forcing to be run last, only a priority level, which can also be the same in multiple plugins, and therefore the fallback on alphabetical by identifier correct? 

I agree this seems like a support nightmare making it configurable at all. The concept of dependency/predecessor resolution for plugins doesn't sound like a bad idea though.

corporategoth, 2020-06-01:
@foosel Octoprint already has a concept of 'Advanced options' - or options that give you enough rope to hang yourself.  Let this be an advanced option under Plugins.  Because otherwise, there is, right now, NO way for a user to control this ordering.

And relying on plugin developers to handle this means that each plugin developer needs to know about all other possible plugins that might use the same hook, and which they should be before or after.  There is no possible way to do this.  You might be able to give plugin developers the way to have hints like 'first', 'neat the beginning', 'anywhere', 'near the end' and 'last', or specify a number, but if two plugins use the same number, and/or there IS a problem with that order, there should be a way for the user to break that tie or override the developer who, possibly, had never even HEARD of the other plugin that is causing problems (or excess processing) due to an ordering issue.