OctoPrint/OctoPrint #3587: [Request] Use Swagger to document the REST API

vringar, 2020-06-02:
**Is your feature request related to a problem? Please describe.**
I want document REST API documentation using OpenAPI so that people can generate stubs for the APIs and explore them interactively.

**Describe the solution you'd like**
This issue will probably require multiple PRs as there are the following things to implement:
- [ ] Generate the OpenAPI definitions from the docstrings on the methods themselves using [flassger](https://github.com/flasgger/flasgger)
- [ ] Generate the documentation from the OpenAPI schemas using [sphinxcontrib-openapi](https://sphinxcontrib-openapi.readthedocs.io/) or [sphinxcontrib-redoc](https://sphinxcontrib-redoc.readthedocs.io/en/stable/) 
- [ ] Removing the hand crafted docs after validating that the new docs aren't lacking any important information

**Additional steps**
- [ ] Figure out how to document the websockets
- [ ] Figure out how to document currently undocumented tornado APIs

**Describe alternatives you've considered**
Handwriting the OpenAPI definitions by hand and have them exist along side the handwritten docs and the code

**Additional context**
I initially proposed this idea in the [community forum](https://community.octoprint.org/t/idea-use-swagger-to-document-the-rest-api/20582)

cp2004, 2020-10-19:
The comment in #3588 makes me think this should also be closed...? Or is the idea still there, but planned implementation not working? Quoting https://github.com/OctoPrint/OctoPrint/pull/3588#issuecomment-703839726
> As far as I understood the framework it only parsed code that had an @api.route annotation, so this won't be possible.

vringar, 2020-10-20:
I mostly gave up on this since I underestimated the time it takes to write JSON Schema.
Reading  @foosel 's comment above mine this was only a suggestion I got way too hung up on.
If she is okay with it, I think the schemas could live as separate files in the tree for the initial implementation. And one could file an issue with flassger folx to ask for guidance on alternatives. These alternatives should then be presented to Gina and she would make the final call.
I still think an OpenAPI documentation is useful and would be willing to help somebody get up to speed should they decide to pick up this issue.