OctoPrint/OctoPrint #3589: [Request] "Update When Print Ends" option

KenLucke, 2020-06-02:
In addition to the "Ignore" and "Update Now" (which does not work while printing, of course) options for software and plugin updates, it would be nice to have a "Update When Print Ends" option that would show up if the dialog is presented during a print.  Clicking that option would wait until the current print job finishes, then do the updates and its restart, after which it would notify the user of the updates installed with an additional dialog.

combs, 2020-10-18:
I was just coming to open a similar feature request. This would fit nicely with how I, at least, use OctoPrint! Thank you!

foosel, 2021-11-23:
Just realized this is actually a duplicate of #3320, but this one has seen more upvotes, so I'm keeping it open and will close #3320 instead.

foosel, 2022-02-25:
Implemented by @jneilliii in #4364 and will be part of 1.8.0