OctoPrint/OctoPrint #3591: Octoprint Crashes with Filament Sensor Plugin

cfelicio, 2020-06-06:
Not sure if this belongs here or with the filament runout plugin. If the latter, let me know and I'll close. Somehow Octoprint is crashing when my filament runout sensor activates:

[image]

I'm running the latest Octoprint with the latest Octopi (1.4.0 / 0.17.0). Steps to reproduce:

1 - Start print
2 - Let filament run out
3 - Octoprint executes pause without an issue
4 - Replace filament
5 - Resume the print via Octoprint -> Print head moves, and Octoprint intermediately crashes afterwards. Print head stops.

I did a separate try of executing the pause / resume without triggering the plugin / out of filament, and there are no crashes. See attached logs, not much information sadly :(

[logs.zip](https://github.com/OctoPrint/OctoPrint/files/4739129/logs.zip)

cfelicio, 2020-06-06:
Oh, also want to point out this plugin was not updated since the release in November / 2019, and it was working fine back then with 1.3.12. I did not change the plugin configuration as far as I remember.

https://github.com/RomRider/Octoprint-Filament-Revolutions/releases

closed 2020-06-06 by cfelicio

tedder, 2020-06-06:
You closed it, I assume you are contacting the plugin author?

cfelicio, 2020-06-06:
Looks like the plugin is dead in terms of support. Since I'm using klipper, I removed the plugin and will try to use the firmware to handle the sensor instead.