OctoPrint/OctoPrint #3593: Documentation for protocol_temperatures_received_hook is missing example code

b-morgan, 2020-06-07:
The documentation for https://docs.octoprint.org/en/master/plugins/hooks.html#protocol_temperatures_received_hook shows:

**Example**

The following example shows how to filter out actual temperatures that are outside a sane range of 1°C to 300°C.

The actual example code is missing.

jneilliii, 2020-06-07:
Good catch @b-morgan, for reference it is there in the previous release branch, see here.

https://docs.octoprint.org/en/1.3.12/plugins/hooks.html#protocol_temperatures_received_hook

And another example can be found here that will allow for adding additional data once the next release of OctoPrint comes out and my plugin is installed.

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/master/test%20plotly%20graph.py

foosel, 2020-06-08:
It's not just that example, apparently all examples pulled in from the example repo aren't present in the 1.4.0 doc build. Yay...

foosel, 2020-06-08:
That has been fixed by 379c969a9f06aa5c1720086afc843a26f8ab0108 and a819b5d2ca2443f3f58532e8525b349877a87d06.

Apparently something changed in Sphinx enough that the custom extension I built to pull in some of the examples from a repo stopped working - and sadly silently.

closed 2020-06-08 by foosel