OctoPrint/OctoPrint #3594: [Firefox Extension Inccompatibility] Zooming via mouse or touchpad in gcode viewer no longer works

KenLucke, 2020-06-07:
I was about to report this as a OP bug, but when I got down to the Browser Being Used part, I decided to check it in Safari (I normally use Firefox), and it went away.  

Checking further, it seems that it is an incompatibility with one single plugin in Firefox, called "NoProfile," which "Blocks functions used for psychological profiling and tracking," and which I had recently re-enabled  

Removing or disabling that extension in Firefox returns OctoPrint to its normal function.

I am only reporting it here so that you may note it for future reference if someone reports it as a bug.

The following details the incompatibility (I wouldn't go so far as to call it a "bug" on the OP side of things).

#### What were you doing?
On the Gcode Viewer tab, attempting to zoom in or out on the model using either the touchpad gestures or using the mouse scroll wheel (or key-mouse combo) no longer work, as did previously.  clicking "Center Viewport on Model" or "Zoom In on Model" function as previously (albeit with their limitations), and dragging the model around in the "viewport" works, but trying to actually manually zoom in or out no longer works.

1. Load a model so that it shows inn the gcode viewer
2. attempt to zoom in or out using the mouse or touchpad

All files are affected

#### What did you expect to happen?
That the zooming function worked as previously

#### What happened instead?
It didn't

#### Did the same happen when running OctoPrint in safe mode?
Yes.  Entirely repeatable in normal AND safe mode

#### Version of OctoPrint
1.4.0 on OctoPi 0.17.0

#### Operating System running OctoPrint
Mac OS X 10.15.5

#### Browser and version of browser, operating system running browser
Firefox v77.0.1 ONLY when running extension called "NoProfile"

foosel, 2020-06-08:
I took a look at the source of the extension in question, and it works by explicitly unbinding all drag and zoom events on all elements in a page. So ... yeah. This extension nukes some functionality critical to the proper working of the gcode viewer. If you want to use the viewer, don't use that extension.

Closing as - as you also noticed - it ain't an issue with OctoPrint but rather the quite draconian extension. If someone falls over this via google the information will still be there.

closed 2020-06-08 by foosel

KenLucke, 2020-06-08:
Thanks for confirming, and as I said, it was just for the record.