OctoPrint/OctoPrint #3595: [Request] Addditional Gcode Viewer Controls

KenLucke, 2020-06-07:
Because the gcode viewer's "Center viewport on model" and "Zoom in on model" apparently do their position and size calculations based upon the full width and depth of the model throughout all layers, it throws the centering and size of the model off because of the purge line that I start every print with along the edge of the buildplate.  Thus, even if the model is only, say, a  centered 20mm calibration cube, it includes my 200mm long purge line 5mm in from the edge of my 310mm build plate, and calculates a model 200mm x 160mm in size, and "zooms" or "centers" on that calculated model.

My proposal:  Add a couple of dropdowns for the zoom and center controls:
```
[] Center viewport on: [ model  ]
                        [ layer  ]
[] Zoom in on: [ model  ]
                [ layer  ]
```

OR a couple of radio buttons to do the same:
```
[] Center viewport on:  O Model  O Layer
[] Zoom in on:  O Model  O Layer
```

ManuelMcLure, 2020-06-08:
Another (maybe easier to implement) possibility would be to add a setting (maybe via a plugin) that would tell the gcode viewer to ignore anything in the gcode file before (a) specific comment(s) that could be added to the slicer's start gcode (or which the slicer generates already). If none of the understood comments are in the file then it would behave as it does now.

KenLucke, 2020-06-08:
> Another (maybe easier to implement) possibility would be to add a setting (maybe via a plugin) that would tell the gcode viewer to ignore anything in the gcode file before (a) specific comment(s) [...]

No, I'd rather it be capable of focusing on a particular layer (and the previous/next layer as it does now) so that if a print, say, starts out the first few layers in bed center, but as it gets higher, its center of the printed layer moves off to the front right, back center, left rear, or wherever, the viewport center focus stays on it, if that's what you choose in the options.

foosel, 2021-01-19:
This has been implemented by #3880 and will be part of 1.6.0.

KenLucke, 2021-01-19:
Awesome.  Thank you, you Devs are great.

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel

richfelker, 2021-12-03:
This change broke the ability to follow how layers stack on each other while scrolling in Z, since it now recenters to each layer. Could it please be made optional to restore the old behavior?